robotology-legacy/wysiwyd #96: Circular dependencies in reactiveLayer

Tobias-Fischer, 2016-04-04:
Hi @clement-moulin-frier @jypuigbo @sockchinglow,

There seem to be several circular dependencies in your modules. This issue is meant to keep track that we need to tackle them. I do understand that it's too rushed for year 2, but it should be done for year 3.

One example:
`behaviorManager` wants to connect to `ears` within the individual behaviors, but ears itself wants to connect to `/BehaviorManager/trigger:i`.

I also see that the `attach` method in several of your modules is executed in the beginning / middle of your `configure` method. This is bad practice, as it should only respond to RPC requests when the initialization of the module is fully done.

One impact I've seen is that `allostatic_plot.py` connects to `behaviorManager` before it's fully initialized, and `behaviorManager` thus returns an incomplete list of behaviors, which in turn leads to issues in the plotter ..

Another problem are the highly duplicated configuration files of your modules. One behavior should only be listed once in one single configuration file, and not multiple times for multiple modules.

Best,
Tobi

clement-moulin-frier, 2016-04-04:
Hi @Tobias-Fischer ,

Yes, refactoring the port connection policy so that it is done externally through yarpmanager is the first thing we'll do in year 3. In the meanwhile please stick to the order we defined for running the modules, which is `ears` -> `sensationManager` -> `homeostasis` -> `behaviorManager` -> `proactiveTagging` -> `allostaticController` -> `allostatic_plot` ( @jypuigbo , please confirm this order is the right one).

> Another problem are the highly duplicated configuration files of your modules. One behavior should only be listed once in one single configuration file, and not multiple times for multiple modules.

Can you please precise? I don't see any duplication in the config files. Note that drives in the `allostaticController` config file and behaviors in the `behaviorManager` config file are two different things: drives define the drive dynamic (in particular according to received sensation) whereas behavior define what has to be executed by the robot when out of the comfort zone.

pattacini, 2016-04-04:
Enforcing an order in launching application modules is very bad... :disappointed:

pattacini, 2016-04-04:
Btw, we definitely need a workaround for this: can you edit the main application xml in order to enforce that order using the `dependency` option?

pattacini, 2016-04-04:
I mean it's very bad if we are forced to comply with that order "_manually_" at launch; but if it is automatically handled it will be a lot less painful.

Tobias-Fischer, 2016-04-04:
I think the problem with that is that the dependencies are circular. So if you ask `behaviorManager` to wait for `ears`, and internally `ears` waits for `behaviorManager` to become alive, the dependencies in the XML will never let you launch all modules.

@maxime-petit and I currently launch all the UPF-related modules at the same time with the XML file, and it works fine. Thus, I suggest not to change anything for now.

Best,
Tobi

pattacini, 2016-04-04:
Uhm then everything is working only by chance... because the current timing allows us to launch those modules at once. However, we cannot leave things as they are @clement-moulin-frier @jypuigbo it's very risky.

Tobias-Fischer, 2016-04-04:
Fully agree with that ..

jypuigbo, 2016-04-04:
There is no longer a predefined order to launch the modules (since Lyon IM), as all the circular dependencies are handled in the right order (no need for superstitions anymore :) ). The only problem comes with the attached port of `behaviorManager` WHEN `allostaticPloter.py` comes into play, as @Tobias-Fischer mentioned and we also identified by the end of last week. (And maybe with `allostaticPloter.py` and other modules that we might not have identified yet..)

(It's really helpful that you pointed out that the attach method is in the middle of the config file, and I'll deal with this asap, so that it doesn't give trouble for the demo. )

For the rest of the reactive layer (meaning exactly: `ears`, `sensationManager`, `proactiveTagging`, `homeostasis`, `allostaticController` and `behaviorManager`), the only _rule_ is: **always stop** all of them, and **always close** all of them. The reason is: there are so many interdependencies between them, that would probably be too expensive to check all the connections at running time. 

Still we know (from Lyon's IM) that there's another bug within `ReactiveLayer` that some of the modules are not closing normally for having loops within the update methods. We are trying to deal with this too, as it conflicts with the previous rule to close/open all the `RL` modules at once. This we'll try to sort it out for Y2. 

Still, this doesn't mean we'll not try to get rid of the circularity, I agree with , just that it's not a problem for Y2 anymore (except for the `allostatic_ploter.py`, that we'll check along the week. 

Thanks for the feedback! I hope this clarifies a bit the issue.

Tobias-Fischer, 2016-04-04:
Sounds good to me.
Regarding the `allostatic_plot.py`: Would it be sufficient to move the call to the `attach` method to the end of `configure` of `behaviorManager`? I think this would sort out the problem that `allostatic_plot` receives an incomplete set of behaviors, as it would just not open because it cannot connect to the RPC port of `behaviorManager`. This in turn would not be a problem because we could add `allostatic_plot.py` along with its dependency on `behaviorManager` in the XML script (which we need to do anyways).

pattacini, 2016-04-04:
Good plan @jypuigbo :+1:

Tobias-Fischer, 2016-04-07:
I just tried moving the `attach` method of `behaviorManager` and it does not work because of the circular dependency with `ears`.

Tobias-Fischer, 2016-04-08:
I thought about another easy workaround. Should only take ten minutes.

jypuigbo, 2016-04-08:
What I did by now, was setting a flag at the end of the config file, so `allostatic_plot.py` knows when `behaviorManager` has finished the configuration. 

Is it sufficient, for the review?

Tobias-Fischer, 2016-04-08:
I think the solution I thought of is more general for the long-term. Let me try whether it works; if not, the solution with the flag will be okay.

Tobias-Fischer, 2016-04-08:
Hi @jypuigbo,
Please have a look at https://github.com/robotology/wysiwyd/commit/88bb30c293466b15182cf72226d3833a46db25cc

I think this makes more sense. Now, `behaviorManager` tries to connect to `ears` rather than the other way around. `behaviorManager` is thus no dependency of `ears` anymore.

I've updated the XML script accordingly. @gregoire-pointeau: Please update the script in Lyon.

closed 2016-04-08 by Tobias-Fischer