robotology-legacy/wysiwyd #97: [iol2opc] remove_all does not remove objects which are not present

Tobias-Fischer, 2016-04-04:
Hi @pattacini,
sending `remove_all` to `iol2opc` does not delete objects which are not present in the `OPC`.

Would be great if we can have a look to fix this :).

Best, Tobi

pattacini, 2016-04-04:
Hi @Tobias-Fischer 

The intended behavior of `remove_all` is to clear all the objects in the `classifier` so as clear the content of the `internal database` and the copies of those objects in the `opc`. It has not been designed to handle different objects.

I might not get the point...

Tobias-Fischer, 2016-04-04:
Hi @pattacini,
Sorry, I think I wasn't clear. With "objects which are not present in the `OPC`" I actually meant "objects which are present in the OPC and have the property `present==0.0` set".
I think the `remove_all` method does not loop over objects which are "dead".

Best, Tobi

pattacini, 2016-04-04:
Ok then, I'll look at it :+1:

pattacini, 2016-04-04:
I don't have a free time slot to test it on the robot but I can give the following guess:

1. When `iol2opc` puts [`m_present=0.0`] (https://github.com/robotology/wysiwyd/blob/master/main/src/modules/systemVisual/iol2opc/src/module.cpp#L739) declaring an object _dead_, it does not remove the object from the database **`db`**.
2. When `remove_all` service gets executed, `iol2opc` [cycles on all objects](https://github.com/robotology/wysiwyd/blob/master/main/src/modules/systemVisual/iol2opc/src/module.cpp#L1176) available in **`db`**.

Given (1) and (2), I'd say that the _dead objects_ should be considered in the removal loop...

Anyway, I'll try to test it asap...

Tobias-Fischer, 2016-04-04:
Hmmm .. We'll check this again whether it happens all the time or whether it was a one-off issue. Thanks for looking into it though!

pattacini, 2016-04-06:
Hi @Tobias-Fischer 

I've just given a go to the pipeline and it seems to me it's working fine in this respect.

Tobias-Fischer, 2016-04-06:
Okay. Might have been a one time problem then. Thanks a lot!

closed 2016-04-06 by Tobias-Fischer

pattacini, 2016-04-17:
See #115 for the fix.