robotology-legacy/wysiwyd #102: Tagging: If only one unknown object, use this one

Tobias-Fischer, 2016-04-05:
If there is only one unknown object in the scene, and the human asks to "point the X" where X is an unknown object, the iCub can assume that it is the only unknown object in the scene. Thus, the iCub should point the object and ask "whether this is the X".

clement-moulin-frier, 2016-04-05:
This is a nice feature to have indeed.
Just a note for Year 3: This is typically a contextual layer functionality (reasoning), so we will certainly have to make some restructuring after the review meeting so that we stay coherent with the DAC framework. It would be actually interesting to know when that kind of reasoning is acquired during infant development.
But for the review it's fine to have it directly in `proactiveTagging`.

Tobias-Fischer, 2016-04-05:
Yes indeed. I suggest we should look into `proactiveTagging` anyways to make it nicer from a software engineering point of view, which should also help to make it more DAC-compatible.

maxime-petit, 2016-04-07:
Fixed in https://github.com/robotology/wysiwyd/commit/27ebc3b514e583d2434e721f6aef1b4cb6cec800

closed 2016-04-07 by maxime-petit