robotology-legacy/wysiwyd #105: narrativeHandler: look for the last X instances to extract stories

maxime-petit, 2016-04-05:
Instead of using the instantStart which depend on the ABM used, the range of ABM to be checked should use relative boundaries (e.g. [max-X:max]), and loaded from conf file.

This is to be able to manage how much time the ABM is querried at startup by narrative: it can be quite some time if you go through a large portion of ABM, and other modules might want to acces ABM in the meantime

./Max

gregoire-pointeau, 2016-04-06:
A first update has been made to reduce considerably the number of request and the display.
@maxime-petit can you try it and tell me if the time is still to long.
I'll continue on the max*min later.

gregoire-pointeau, 2016-04-06:
commit: https://github.com/robotology/wysiwyd/commit/f924a4f6345761759947184faaaead299fea7b7a
have added a min max (`instanceStart`, `instanceStop`) in the conf file.
However previous commit should have increase speed.
Please try on all your DB just to check if there is a significant improvement.

Tobias-Fischer, 2016-04-06:
Hi @gregoire-pointeau,
Rather than defining absolute values, is it possible to define relative ones? We have dozens of new instances every day, which means that every day you would need to change the absolute value. What @maxime-petit wants is a relative value, i.e. take the last X instances instead of take instances Y to Z which was now implemented.

gregoire-pointeau, 2016-04-06:
Ok,.
In the mean time, can you confirm that there is however an improvement in term of speed for starting?

gregoire-pointeau, 2016-04-06:
There are now two more parameters:
`researchWindows`   default:   `1`      1 is window (min to max) while 0 is n-back
`nBackSize`   default `200`     size of the n-last instance to check.

closed 2016-04-06 by gregoire-pointeau

Tobias-Fischer, 2016-04-06:
Did you forget to commit?

gregoire-pointeau, 2016-04-06:
...
maybe ...