robotology-legacy/wysiwyd #106: [agentDetector] Agent's presence remains 0.0

sockchinglow, 2016-04-05:
Our `agentDetector` does not update the presence of any partner even when there is a skeleton in the viewer i.e. partner's `presence = 0.0` regardless of detection.

As @Tobias-Fischer has suggested, we have checked that our version for `agentDetector` is updated to match the latest one from the `year-2-demo` branch, as is its config file. Also, we have no config file in the .local folder to mess with this.

Has anyone other than Imperial tried using any methods which require the presence of a partner, or the `lookAtAgent()` method itself? A simple check would be that the robot has looked at the partner intermittently while running the `pointing` (knowledge expression) behaviour during one of your tests.

maxime-petit, 2016-04-05:
Hi,

1) Do you have the skeleton appearing in the window?
2) The joints position in the partner's entity are changing? 

./Max

sockchinglow, 2016-04-05:
Hey Max,

1) yes, the skeleton appears and
2) yes, they change according to the partner's posture.

jypuigbo, 2016-04-06:
Could the option `dThresholdDisparition` be affecting that? Could you check if you have in your config file some value different from the default one? (`3.0`) (it is not in the default config file in `main/apps/humanUnderstanding/agentDetector/conf/`

Thanks

Tobias-Fischer, 2016-04-06:
Hi @jypuigbo,
We use the default configuration file here.
If you want, @maxime-petit and I can have a brief look via teamviewer to see what might be wrong.

Best, Tobi

Tobias-Fischer, 2016-04-07:
Any news @jypuigbo @sockchinglow?

sockchinglow, 2016-04-07:
No improvements so far...I'll send you the teamviewer information via slack

sockchinglow, 2016-04-07:
Thanks to @Tobias-Fischer, we have found the source of the problem - a simple matter of building and installing. Resolved and closed.

closed 2016-04-07 by sockchinglow

Tobias-Fischer, 2016-04-07:
@robotology/wysiwyd-developers: As we have seen the same issue in Lyon and it took us a long while to sort it out, I want to recap what happened. On one of the machines @sockchinglow was using, someone did a `sudo make install` a while back and forgot about it. Other people where expecting that the binaries in the build folder are used when executing modules. However, the binaries in `/usr/local/bin` have priority in their `PATH` variable.
My recommendation: On all of your machines, have the same policy of either running `make install` or just `make`. If you have different policies on different machines, it will likely cause problems one day or another.
Furthermore, setup the `update-software.py` on all your machines: https://github.com/robotology/wysiwyd/tree/year-2-demo/admin/scripts/update-software

Best, Tobi

pattacini, 2016-04-07:
Perhaps it'd be worth becoming a bit more maniac - as I am - and every once in a while do a random check of the system status.

For example, just after building the project you could verify if the outcome of the following command

```sh
$ which binary-name
```

matches the expected path...

pattacini, 2016-04-07:
Other useful things a maniac does:

- call `yarp clean --timeout 0.1` to see if there're stale ports hanging off.
- check if `ps aux | grep yarprun` tells that there are only 2 instances of `yarprun` on that machine (yes they must be 2).

maxime-petit, 2016-04-07:
Other thing related: always put `$YARP_DIR`,`$ICUB_DIR` and `$WYSIWYD_DIR`before anything in the `$PATH`. In that case, your repo will always have priority over the /usr/local/bin. However you will look for it if for some reason you are not building the executable anymore so be careful anyway