robotology-legacy/wysiwyd #109: `iCub->say()` failing

gregoire-pointeau, 2016-04-07:
Hello,
I use the new speech `svox` and I have the impression that the subsystem speech cannot get a response from the new speech module when launching a `say` command.

In several modules that I tried, I cannot get a response from the speech which either block the module (if you're waiting for a response) or doesn't allow you to send more than one sentence by module.
A simple test is simply to add:
`iCub->say("sentence 1");`
`iCub->say("sentence 2");`
at the beginning of a module and check if the iCub is saying it.

Tobias-Fischer, 2016-04-07:
Cannot confirm this. Working fine here. Do you follow the instructions given in https://github.com/robotology/speech/pull/11 ? Especially, does the "test" in these instructions work? If so, `iCub->say` should work.

gregoire-pointeau, 2016-04-07:
The test give in instructions works fine. It just in the modules that I have problems.
I'll double check the connections and my `xml` file, and I'll tell you.

pattacini, 2016-04-07:
Hi @gregoire-pointeau 

With the new speech synthesizer you have to make a further connection:

```
$ yarp connect /iSpeak/speech-dev/rpc /icub/speech:rpc
```

gregoire-pointeau, 2016-04-07:
Ok, I solved it, it was coming from the carrier for the connection. It should be `tcp` and not `udp`.
Solved.

closed 2016-04-07 by gregoire-pointeau

pattacini, 2016-04-07:
Uhm... `udp` is obviously less reliable than `tcp` but now I'm wondering whether with `udp` protocol you were getting no utterance systematically. If this was the case, I'm afraid your network in on the "edge" of correct operation.

gregoire-pointeau, 2016-04-07:
by default, the `connect` command in yarp, is using which?

pattacini, 2016-04-07:
By default it's using `tcp`.
You can check the connection's type by querying `yarp ping /port-name`.

gregoire-pointeau, 2016-04-07:
Ok, so this make sense.
When I was connecting by hand it was working, not through `yarpmanager` that was set on `udp` (changed now)

pattacini, 2016-04-07:
My point is that there must be a very high usage of the network if, before, you didn't manage to get any utterance at all from `iSpeak` via `udp` connection, regardless of the number of attempts you made.

gregoire-pointeau, 2016-04-07:
Yes indeed, I can let `iSpeak` running, and launch several times my module, it will only say the first sentence each time.
I'm trying it with a very low number of modules running actually (5).

Tobias-Fischer, 2016-04-07:
Can you confirm that everything is running fine using `festival` instead of `svox`?

gregoire-pointeau, 2016-04-07:
Yes. It is.