robotology-legacy/wysiwyd #118: Kinect Calibration with several objects

maxime-petit, 2016-04-25:
Hi,

The stereo-vision has better result (stability, localisation of object, size of objects, etc.) in a non-plannar environment, e.g. when several objects are on the table. 

However, the current Kinect Calibration require a single object: the situation is not optimal, on top of which the object has to be put quite far from the icub because the kinect need to see it also. 

We might want to improve this calibration by allowing several objects presents in the scene, but using a specific object with a precise name to "click" during the calibration.

./Max

maxime-petit, 2016-04-25:
FYI: The current "Jedi Trick" is to put very large objects around that are not recognize as object by IOL (so we have only one object present in the OPC but in a non-plannar scene)

Tobias-Fischer, 2016-04-25:
Hi Max,

I don't get the logic behind this. Albeit the SFM needs a non planar scene at the time of calibration, the performance should not be affected by the number of objects.

So I don't see where a performance improvement would happen if there are multiple objects on the table when calibrating the Kinect to the robot coordinate frame.

I guess the problem on our robot might be a bad fine calibration of neck, head, eyes, ...

Best,
Tobi
PS: @pattacini what do you think?

pattacini, 2016-04-25:
Hi @Tobias-Fischer 

Unfortunately, with only one object (often of small dimensions) lying on a complete flat and indistinguishable surface, `SFM` performs quite poorly giving flickering results, which are due to the fact that many `(0,0,0)` invalid points are returned.

pattacini, 2016-04-25:
Non-planar scenes are required mostly during `SFM` calibration but also during run-time operations in order to allow for correct features matching.

Tobias-Fischer, 2016-04-25:
Ahhh I wasn't aware of this. Sorry! I'm happy to implement the feature Max requests with multiple objects when I have a free minute, it should not take much longer than that.

Best, Tobi

Tobias-Fischer, 2016-05-06:
Just a quick update: @maxime-petit and I didn't get around to test this yet, we'll try and do it asap.

closed 2016-05-06 by Tobias-Fischer