robotology-legacy/wysiwyd #119: SAM action recognition should be single module running

Tobias-Fischer, 2016-04-27:
Hi @dcam0050,

Please change SAM such that it does not spawn new windows. If you need to have separate processes, please launch them as threads rather than launching new processes in `xterm` windows.
For the visualization, you could have a separate window purely for this task retrieving the data from the main process.
I am strongly against the hack we used in Lyon for the Year 2 demo, this was only good for a very temporary solution.

Best,
Tobi
/cc @pattacini for software engineering questions

Tobias-Fischer, 2016-04-27:
Furthermore, a pre-trained model should be provided. Training locally is not an option, the module should run out of the box. You should use `RFModule` to locate the model file (as well as any other configuration file). Hard coded paths are evil ..

Tobias-Fischer, 2016-06-21:
Any news on this @dcam0050?

dcam0050, 2016-06-21:
Working on it should release something in the next 2 weeks.

dcam0050, 2016-07-29:
This is fixed with commit #147

Tobias-Fischer, 2016-07-31:
Will close after merging https://github.com/robotology/wysiwyd/pull/147

Tobias-Fischer, 2016-09-05:
Fixed in #147

closed 2016-09-05 by Tobias-Fischer