robotology-legacy/wysiwyd #120: SAM with dictionnary of force restults verb in the ABM

gregoire-pointeau, 2016-04-27:
`SAM` action recognizer should be able to save recognized action into the ABM automatically.
Each action can have a distinction `force/result` (ie: `drop/fall` `push/moved`) that can be associated in an `.ini` file.

I can supervised the part about the recording in the `ABM`.

Tobias-Fischer, 2016-08-17:
Hi @dcam0050, is this addressed in your PR? Is this still needed @gregoire-pointeau?

gregoire-pointeau, 2016-08-17:
"Needed" I don't know.
It's just for SAM to be able to imagine different point of view.

Tobias-Fischer, 2016-10-03:
@gregoire-pointeau @dcam0050 did you ever talk about this again? Do you think this will happen before the end of the year? Otherwise will close as `wontfix`.

gregoire-pointeau, 2016-10-03:
We talked about sending all open class word from `LRH` to `SAM` but I guess the dictionary won't be set before the end of the year, we can close it I guess.

closed 2016-10-03 by gregoire-pointeau