robotology-legacy/wysiwyd #121: Remove warnings in narrativeHandler

Tobias-Fischer, 2016-04-27:
Hi @gregoire-pointeau,
One of your recent commits introduced the following warnings:
```
/home/icub/robot/wysiwyd/main/src/modules/narrativeRelated/narrativeHandler/src/humanNarration.cpp: In member function ‘bool narrativeHandler::narrationToMeaning()’:
/home/icub/robot/wysiwyd/main/src/modules/narrativeRelated/narrativeHandler/src/humanNarration.cpp:423:29: warning: unused variable ‘p’ [-Wunused-variable]
                         int p = 0;
                             ^
/home/icub/robot/wysiwyd/main/src/modules/narrativeRelated/narrativeHandler/src/humanNarration.cpp: In member function ‘evtStory narrativeHandler::adaptMeaning(evtStory&)’:
/home/icub/robot/wysiwyd/main/src/modules/narrativeRelated/narrativeHandler/src/humanNarration.cpp:476:28: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
         if (pos != string::npos){
                            ^
/home/icub/robot/wysiwyd/main/src/modules/narrativeRelated/narrativeHandler/src/humanNarration.cpp:488:65: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
     for (unsigned int iRel = 0; iRel < evtInput.bRelations.size(); iRel++){
                                                                 ^
/home/icub/robot/wysiwyd/main/src/modules/narrativeRelated/narrativeHandler/src/humanNarration.cpp:493:55: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
         for (unsigned int iSub = 0; iSub < bTemp.size(); iSub++){
                                                       ^
/home/icub/robot/wysiwyd/main/src/modules/narrativeRelated/narrativeHandler/src/humanNarration.cpp: In member function ‘void narrativeHandler::imagineStory(story&)’:
/home/icub/robot/wysiwyd/main/src/modules/narrativeRelated/narrativeHandler/src/humanNarration.cpp:582:63: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
         for (unsigned int iRel = 0; iRel < bAllInstances.size(); iRel++){
                                                               ^
/home/icub/robot/wysiwyd/main/src/modules/narrativeRelated/narrativeHandler/src/humanNarration.cpp:586:59: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
             for (unsigned int iSub = 0; iSub < bTemp.size(); iSub++){
                                                           ^
/home/icub/robot/wysiwyd/main/src/modules/narrativeRelated/narrativeHandler/src/humanNarration.cpp:619:53: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
         for (unsigned int ii = 0; ii < bNewOpc.size(); ii++){
                                                     ^
/home/icub/robot/wysiwyd/main/src/modules/narrativeRelated/narrativeHandler/src/humanNarration.cpp:671:57: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
             for (unsigned int ii = 0; ii < bNewOpc.size(); ii++){
                                                         ^
/home/icub/robot/wysiwyd/main/src/modules/narrativeRelated/narrativeHandler/src/humanNarration.cpp:688:28: warning: unused variable ‘prev’ [-Wunused-variable]
                     size_t prev = 0;
```

As I've recently removed all warnings in WYSIWYD, it would be nice if you get rid of these ones :). Please let me know if you don't know how to resolve some of them.

Best,
Tobi

gregoire-pointeau, 2016-04-29:
I'll resolved it, I need to reset my setting of error in `visual` because I don't have them.

Tobias-Fischer, 2016-05-01:
Still some left after the initial fixes (https://github.com/robotology/wysiwyd/commit/cd155cee1ce2388c34edc245b9c5e9a60ec3a343 thanks @gregoire-pointeau):
```
/home/tobias/robot/wysiwyd/main/src/modules/narrativeRelated/narrativeHandler/src/humanNarration.cpp: In member function ‘evtStory narrativeHandler::adaptMeaning(evtStory&)’:
/home/tobias/robot/wysiwyd/main/src/modules/narrativeRelated/narrativeHandler/src/humanNarration.cpp:465:33: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
             for (int jj = 0; jj < comparator.size(); jj++){
                                 ^
/home/tobias/robot/wysiwyd/main/src/modules/narrativeRelated/narrativeHandler/src/humanNarration.cpp: In member function ‘void narrativeHandler::imagineStory(story&)’:
/home/tobias/robot/wysiwyd/main/src/modules/narrativeRelated/narrativeHandler/src/humanNarration.cpp:559:37: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
                 for (int jj = 0; jj < comparator.size(); jj++){
                                     ^
/home/tobias/robot/wysiwyd/main/src/modules/narrativeRelated/narrativeHandler/src/humanNarration.cpp:589:38: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
         for (unsigned int ii = 0; ii < bNewOpc.size(); ii++){
```

gregoire-pointeau, 2016-05-02:
To be honest I don't understand why sometime you cannot compare a `.size()` with either a `signed` or `unsigned int` ... I'll investigate on it.

pattacini, 2016-05-02:
@gregoire-pointeau it really depends on the return type of `.size()`.

For instance:

- `yarp::os::Bottle::size()` returns a `int`, thus you should compare it with `int` to avoid generating a warning.
- `std::vector<>::size()` returns a `size_t` value, which is basically a `unsigned int`. Therefore, you have to behave accordingly.

gregoire-pointeau, 2016-05-02:
Ok, I'll change it, and should I try the new PR system?

pattacini, 2016-05-02:
Yes why not, even if it's a very tiny change: just to exercise yourself :smile:.

Tobias-Fischer, 2016-05-06:
Can be closed as soon as https://github.com/robotology/wysiwyd/tree/inserm-narrative is merged.

pattacini, 2016-05-09:
Fixed via #129.

closed 2016-05-09 by pattacini