robotology-legacy/wysiwyd #122: Error on visualdata table when issuing load in ABM

dcam0050, 2016-04-28:
I have just created a new ABM database by issuing new to autobiographicalMemory via rpc and then issued a load command and received the log below:

command received: load
NOTICE:  drop cascades to 5 other objects
DETAIL:  drop cascades to constraint contentopc_instance_fkey on table contentopc
drop cascades to constraint beliefs_key on table beliefs
drop cascades to constraint contentarg_instance_fkey on table contentarg
drop cascades to constraint drives_instance_fkey on table drives
drop cascades to constraint emotions_instance_fkey on table emotions
NOTICE:  drop cascades to 7 other objects
DETAIL:  drop cascades to constraint entity_instance_fkey on table entity
drop cascades to constraint action_instance_fkey on table action
drop cascades to constraint adjective_instance_fkey on table adjective
drop cascades to constraint object_instance_fkey on table object
drop cascades to constraint rtobject_instance_fkey on table rtobject
drop cascades to constraint agent_instance_fkey on table agent
drop cascades to constraint relation_instance_fkey on table relation
NOTICE:  drop cascades to 5 other objects
DETAIL:  drop cascades to table action
drop cascades to table adjective
drop cascades to table object
drop cascades to table rtobject
drop cascades to table agent
NOTICE:  table "action" does not exist, skipping
NOTICE:  table "adjective" does not exist, skipping
NOTICE:  table "object" does not exist, skipping
NOTICE:  table "rtobject" does not exist, skipping
NOTICE:  table "agent" does not exist, skipping
NOTICE:  drop cascades to constraint spatialdata_instance_fkey on table spatialdata
NOTICE:  drop cascades to 2 other objects
DETAIL:  drop cascades to constraint contextdata_instance_fkey on table contextdata
drop cascades to constraint contextagent_instance_fkey on table contextagent
NOTICE:  drop cascades to constraint timedata_temporal_fkey on table timedata
NOTICE:  drop cascades to constraint behaviordata_instance_fkey on table behaviordata
NOTICE:  drop cascades to 2 other objects
DETAIL:  drop cascades to constraint sharedplanarg_instance_fkey on table sharedplanarg
drop cascades to constraint sharedplandata_instance_fkey on table sharedplandata
NOTICE:  drop cascades to constraint spdataarg_instance_fkey on table spdataarg
NOTICE:  table "visualdata" does not exist, skipping
terminate called after throwing an instance of 'DataBaseError'
  what():  Failed to execute sql: Error: PGRES_FATAL_ERROR : ERROR:  syntax error at or near "("
LINE 1: ... without time zone CONSTRAINT img_pkey PRIMARY KEY("time", i...
 

gregoire-pointeau, 2016-04-28:
cc @Tobias-Fischer @maxime-petit 
I'm not very familiar with the table `visualdata`, maybe you know the proper format of time zone?

maxime-petit, 2016-04-28:
Yop!

Yes, the load method is not updated. I think we solved it when we were in Lyon. 

The visualdata table should be created like this:
`-- Table: public.visualdata

-- DROP TABLE public.visualdata;

CREATE TABLE public.visualdata
(
  "time" timestamp without time zone NOT NULL,
  img_provider_port text NOT NULL,
  instance integer NOT NULL,
  relative_path text,
  img_oid oid,
  augmented text,
  frame_number integer NOT NULL,
  augmented_time timestamp without time zone NOT NULL DEFAULT '2000-01-01 00:00:00'::timestamp without time zone,
  CONSTRAINT img_pkey PRIMARY KEY ("time", img_provider_port, augmented_time),
  CONSTRAINT images_instance_fkey FOREIGN KEY (instance)
      REFERENCES public.main (instance) MATCH SIMPLE
      ON UPDATE NO ACTION ON DELETE NO ACTION
)
WITH (
  OIDS=FALSE
);
ALTER TABLE public.visualdata
  OWNER TO postgres;

-- Index: public.visualdata_instance_time

-- DROP INDEX public.visualdata_instance_time;

CREATE INDEX visualdata_instance_time
  ON public.visualdata
  USING btree
  (instance, "time");
`

I ll check to change the load method in the meantime

maxime-petit, 2016-04-28:
Hum, weird it seems that it is properly configure in `year-2-demo` branch.

Can you send me the exact query sent to ABM please?

Please check if this commit is taken into account on your system: https://github.com/robotology/wysiwyd/commit/a833d516167d32a7b6e27a7ff9fe4ef50cb049fe

Tobias-Fischer, 2016-04-28:
Please try again after applying https://github.com/robotology/wysiwyd/commit/3a566e0a6356a74805098e6f6aee738b754c1888

dcam0050, 2016-04-28:
Will do

dcam0050, 2016-04-28:
I pulled wysiwyd, compiled and ran autobiographicalMemory then sent rpc command 'new' followed by 'load' and I still have the same error

Tobias-Fischer, 2016-04-28:
Try https://github.com/robotology/wysiwyd/commit/a6d07caa96401e180bd9afbb5bb81eea028e0c77 (on the `devel` branch.

closed 2016-04-28 by Tobias-Fischer

dcam0050, 2016-04-28:
Works like a charm!