robotology-legacy/wysiwyd #123: Enforce dependency from OpenCV-3.0.0 or higher

pattacini, 2016-04-28:
I would enforce the dependency from **`opencv-3.x.y`** from now on and thus streamline code and documentation.

Is that ok for you?

cc @Tobias-Fischer

Tobias-Fischer, 2016-04-28:
Fine with me :+1:

pattacini, 2016-04-28:
Good then.
I'll apply this change to master, thus waiting for it.

pattacini, 2016-04-28:
@robotology/wysiwyd-developers 

From 0f0f871 on, `iol2opc` will compile only against **`OpenCV-3.x.y`**.
At the same time, the cmake variable **`WYSIWYD_IOL2OPC_WITH_TRACKING`** is no longer needed.

closed 2016-04-28 by pattacini

pattacini, 2016-04-29:
For completeness: to avoid messing up CI because of the dependency from opencv external modules, I've introduced the cmake variable **`WYSIWYD_BUILD_IOL2OPC`**, which regulates the compilation of `iol2opc`, in particular by disabling it during CI.