robotology-legacy/wysiwyd #125: Create a "Goals" module

clement-moulin-frier, 2016-04-28:
Handling human requests (like *point the banana*) is currently handle in `ears`. Human requests should indeed be transformed into a goal, which is then decomposed as an action plan (e.g. *first ask what is the banana, then push it"). Also we might consider self-generated goals in the near future (i.e. not coming from a human request but directly generated by the robot itself), so the new "Goals" module should be general enough to handle this case. In particular speech recognition should be performed ahead, not in this module.

Choosing what behavior will be eventually triggered (either from the reactive layer or from an action plan at the contextual level) is the responsibility of the `allostaticController` which will contain a drive for realizing a contextual goal (for the time being the `followingOrder` behavior is not linked to a drive).

Tobias-Fischer, 2016-10-27:
@clement-moulin-frier @jypuigbo How far are you guys with this issue?

clement-moulin-frier, 2016-10-28:
It's done but we still have to adapt a few behaviors of the Y2 demo to deal with the new architecture. We are on it. (a few issues to solve before, as explained during the last slack meeting).

Tobias-Fischer, 2016-11-28:
Hey @clement-moulin-frier,
what's the status on this?

clement-moulin-frier, 2016-11-28:
This is done in the planner, we can close

closed 2016-11-28 by clement-moulin-frier