robotology-legacy/wysiwyd #127: Remove AppVeyor patch for icub-main

pattacini, 2016-04-28:
Remove this [**`patch`**](https://github.com/robotology/wysiwyd/blob/master/appveyor.yml#L49-L50) as soon as the new icub-main release will be available.

Tobias-Fischer, 2016-05-24:
Hi @pattacini,
I think we can tackle this now with the new release being available. Also, in `travis` we can get rid of the `--force-yes` if we instead add the public key of the repository.

pattacini, 2016-05-25:
Hi @Tobias-Fischer 

The Windows packages are still to be released, apparently. I don't see them here yet: https://github.com/robotology/icub-main/releases/tag/v1.1.17.

Concerning the `--force-yes` option, we should first address this [comment](https://github.com/robotology/icub-packaging/issues/3#issuecomment-221545962).

pattacini, 2016-05-30:
Hi @Tobias-Fischer 

- Done with the `--force-yes` option: just awaiting confirmation from **travis**.
- Windows packages are now available but I spotted a problem with **Ipopt**: the installer does not unpack **Ipopt** correctly. We ought to wait for another release.

Tobias-Fischer, 2016-05-30:
Sounds good :).
So for now, `appveyor` is still using the old release?

pattacini, 2016-05-30:
Yep ;)

pattacini, 2016-05-31:
d4ef937 causes AppVeyor to fail with the following errors:

```
Error	5	error LNK1169: one or more multiply defined symbols found	C:\dev\wysiwyd\main\build\bin\Release\agentDetector.exe
Error	2	error LNK2005: "public: __thiscall yarp::os::PortReaderBuffer<class yarp::sig::ImageOf<struct yarp::sig::PixelRgb> >::PortReaderBuffer<class yarp::sig::ImageOf<struct yarp::sig::PixelRgb> >(unsigned int)" (??0?$PortReaderBuffer@V?$ImageOf@UPixelRgb@sig@yarp@@@sig@yarp@@@os@yarp@@QAE@I@Z) already defined in kinectWrapper.lib(kinectWrapper_client.obj)	C:\dev\wysiwyd\main\build\src\modules\humanUnderstanding\agentDetector\YARP_dev.lib(YARP_dev.dll)
Error	3	error LNK2005: "public: virtual __thiscall yarp::os::PortReaderBuffer<class yarp::sig::ImageOf<struct yarp::sig::PixelRgb> >::~PortReaderBuffer<class yarp::sig::ImageOf<struct yarp::sig::PixelRgb> >(void)" (??1?$PortReaderBuffer@V?$ImageOf@UPixelRgb@sig@yarp@@@sig@yarp@@@os@yarp@@UAE@XZ) already defined in kinectWrapper.lib(kinectWrapper_client.obj)	C:\dev\wysiwyd\main\build\src\modules\humanUnderstanding\agentDetector\YARP_dev.lib(YARP_dev.dll)
Error	1	error LNK2005: "public: virtual __thiscall yarp::os::TypedReader<class yarp::sig::ImageOf<struct yarp::sig::PixelRgb> >::~TypedReader<class yarp::sig::ImageOf<struct yarp::sig::PixelRgb> >(void)" (??1?$TypedReader@V?$ImageOf@UPixelRgb@sig@yarp@@@sig@yarp@@@os@yarp@@UAE@XZ) already defined in kinectWrapper.lib(kinectWrapper_client.obj)	C:\dev\wysiwyd\main\build\src\modules\humanUnderstanding\agentDetector\YARP_dev.lib(YARP_dev.dll)
Error	4	error LNK2005: "public: void __thiscall yarp::os::PortReaderBuffer<class yarp::sig::ImageOf<struct yarp::sig::PixelRgb> >::detach(void)" (?detach@?$PortReaderBuffer@V?$ImageOf@UPixelRgb@sig@yarp@@@sig@yarp@@@os@yarp@@QAEXXZ) already defined in kinectWrapper.lib(kinectWrapper_client.obj)	C:\dev\wysiwyd\main\build\src\modules\humanUnderstanding\agentDetector\YARP_dev.lib(YARP_dev.dll)
```

Apparently, `kinectWrapper` started conflicting with recent `yarp` code... I'm on it.
@drdanz any clue?

drdanz, 2016-05-31:
It's MSVC stuff and I know nothing about. It's probably related to robotology/yarp@627fde431151bc16f64e462c3f312935616cd193
cc @lornat75

pattacini, 2016-06-03:
Problem partially solved via https://github.com/robotology/yarp/commit/1aeb89b73132e677ba76faa20e6714979ec6c7e9.
Expected new Yarp release.

commit 0137cd2c (2016-06-08): [appveyor] Fix #127
commit 5fa9e9b4 (2016-06-08): [appveyor] Another try to fix #127
closed 2016-06-08 by pattacini