robotology-legacy/wysiwyd #134: babbling module causes errors in yarprobotinterface

Tobias-Fischer, 2016-05-31:
Hi @martz21,
Some recent changes in `yarprobotinterface` cause the `babbling` module to produce hundreds of errors. The babbling however is working fine. I guess somewhere the change of control mode is missing or something like this. The errors look as follows:
```
450691		2134.435622		ERROR		setPositionRaw: skipping command because    joint  2 is not in VOCAB_CM_POSITION_DIRECT mode 
450693		2134.435637		ERROR		velocityMoveRaw: skipping command because    joint  3 is not in VOCAB_CM_VELOCITY mode 
```

It would be great if you have a look in a free minute.

Thanks,
Tobi
PS: Maybe @pattacini knows more about this?

Tobias-Fischer, 2016-05-31:
I had a quick look and it seems like you have to use `VOCAB_CM_POSITION_DIRECT` mode instead of `VOCAB_CM_POSITION` mode. But again @pattacini probably knows better.

pattacini, 2016-05-31:
yep, @Tobias-Fischer is right.
Also, it seems it must be happening the same for controlling in velocity.

Since something like two years ago we drastically changed the way motors can be controlled. The corresponding mode must be enabled explicitly before calling the relative driving routines.

Tobias-Fischer, 2016-05-31:
So just changing `VOCAB_CM_POSITION` into `VOCAB_CM_POSITION_DIRECT` will do the job? I know that @martz21 already explicitly sets the control mode, maybe just not the proper one ..

pattacini, 2016-05-31:
One thing is certainly to set the correct mode; another thing is also to set it at the right time, that is when it's needed. It may happen that the mode is changed several times (even by other modules) and the user code is not always aware of those changes.

Tobias-Fischer, 2016-08-17:
Will close this as a `wontfix`.

closed 2016-08-17 by Tobias-Fischer