robotology-legacy/wysiwyd #138: Linking the language model to Goal module in #125

clement-moulin-frier, 2016-06-01:
so that action plans can be mapped to verbal sentences bidirectionally.

See D5.3 description. Work in collaboration between UPF and INSERM That's more or less the topic of the next WP5 deliverable in close collaboration with inserm)

Tobias-Fischer, 2016-11-28:
Any news on this @clement-moulin-frier @gregoire-pointeau?

gregoire-pointeau, 2016-11-28:
As I said in #137 We understand and produce complex sentences.
I'll see to push it into ears to create the goals.

clement-moulin-frier, 2016-11-28:
@gregoire-pointeau : let's discuss here what the best way of sending goals, together with @sockchinglow and @jypuigbo (sending it to `ears` might not be the best solution since how we handle action plans has changed a bit). Can you please tell us what is the format of the meaning created by `lrh` on such complex sentences?

Tobias-Fischer, 2016-12-14:
Hi @gregoire-pointeau @clement-moulin-frier, what shall we do with this issue?

clement-moulin-frier, 2016-12-14:
If the robot is able to talk about goal execution through action plan I think it is sufficient. It will actually already be bidirectional, since we can also trigger an action plan through language (even if we can't refine an action plan through language, but it's fine I think).
@gregoire-pointeau : is the robot now able to describe the action plan that just happened? If so we can close I think