robotology-legacy/wysiwyd #143: Report on stale branches

pattacini, 2016-06-12:
The following three branches are almost to be considered as _stale_, given that they haven't been touched since more than 2 months:

- [`renaming2`](https://github.com/robotology/wysiwyd/tree/renaming2), whose responsible developer is @jypuigbo 
- [`babblingTest`](https://github.com/robotology/wysiwyd/tree/babblingTest), whose responsible developer is @martz21 
- [`upf-devel-goals`](https://github.com/robotology/wysiwyd/tree/upf-devel-goals), whose responsible developer is @jypuigbo

Can the developers in charge report on their current state, so that we might possibly make some clean up?

jypuigbo, 2016-06-15:
I've been looking at those branches you tagged me recently. `upf-devel-goals` has work in progress (currently I'm trying to rebase `devel` into it to make sure it is updated), we are constructing the contextual layer there.

For `renaming 2`, it was developed by @clement-moulin-frier  during an integration meeting, but I think the changes were not accepted to be added in the demo. We should talk (he is not around today) to see wether we keep it, or what we do with it. We'll try to give an answer before the end of the week.

pattacini, 2016-06-15:
@martz21 any update on `babblingTest`?

martz21, 2016-06-15:
Hi @pattacini ! No updates yet, but I am going to work on it very soon. If it was possible I would like to keep this branch for another bit. Please let me know if that would be an issue.

pattacini, 2016-06-15:
Cool, thanks for the info @martz21.
The branch can certainly stay there.

closed 2016-06-15 by pattacini