robotology-legacy/wysiwyd #144: [wysiwyd] unable to build as 'opencv2/opencv.hpp' cannot be found

sockchinglow, 2016-06-28:
I have installed opencv3.0.0 as instructed in the WYSIWYD readme. The opencv installation goes fine, but when I try to build WYSIWYD there will be the following error:
`error C1083: Cannot open include file: 'opencv/opencv.hpp': No such file or directory`

This is found in the projects `cvxFiber`, `cvzCore`, `cvzHierarchyXmlBuilder` and `cvzStackExample`. However in `cvzVisualHierarchy`, there were several other errors, mostly syntax, which is also strange. Has anyone any idea how this can be resolved?

Tobias-Fischer, 2016-06-28:
1) Try removing your `build` folder, and try again from scratch.
2) Please post the output of `cat CMakeCache.txt | grep OpenCV`

sockchinglow, 2016-06-28:
I have removed the `build` folder and rebuilt it...several times. To the same result. And I'm using windows, so I'm not sure which file `cat CMakeCache.txt | grep OpenCV` should yield, but I have attached the `CMakeCache.txt` in opencv/build.
[CMakeCache.txt](https://github.com/robotology/wysiwyd/files/337033/CMakeCache.txt)

Tobias-Fischer, 2016-06-28:
I meant the `CMakeCache.txt` in `wysiwyd/main/build` ..
As it's Windows, I won't be able to help. Maybe @pattacini or @maxime-petit can have a look.

sockchinglow, 2016-06-28:
This is the [CMakeCache.txt](https://github.com/robotology/wysiwyd/files/337038/CMakeCache.txt) in `wysiwyd/main/build`.

Tobias-Fischer, 2016-06-28:
I assume that this is due to not having `OpenCV` in the `include_directories` and `target_link_libraries` in the modules you mentioned, although they indirectly include `opencv/opencv.hpp` (they include `cvz/gui/all.h` which includes `GuiNN.h` which includes `cvz/core/CvzNN.h` which includes `opencv2/opencv.hpp`. 

1) Can you remove `opencv2/opencv.hpp` from ``cvz/core/CvzNN.h` and see whether it compiles?
2) Can you try adding `${OpenCV_INCLUDE_DIRS}` and `${OpenCV_LIBRARIES}` respectively in the `CMakeLists.txt` files of the modules you refer to, and see whether this solves the issue?

sockchinglow, 2016-06-28:
It still doesn't work; I have attached one of the `CMakeLists.txt` (from `cvzHierarchyXmlBuilder`) for reference as to how I added the `${OpenCV_INCLUDE_DIRS}` and `${OpenCV_LIBRARIES}`.

Tobias-Fischer, 2016-06-28:
There is no reference to `OpenCV` in your `CMakeLists.txt` at all ..
Did you try solution 1)?
If this doesn't help, I'm afraid @pattacini or @maxime-petit have to kick in, as I have no clue about compiling on Windows.

sockchinglow, 2016-06-28:
I added it into the `include_directories` of the `CMakeLists.txt`. Solution 1) did not help either :/ it's all very strange because I did manage to build it before. I thought it would be something that I changed, so I pulled the entire `wysiwyd` folder from github and tried to recompile from there but it still didn't work.

maxime-petit, 2016-06-28:
Do you have the `opencv.hpp` file in `C:/Users/sock/robotology/opencv/build/opencv2` ?

sockchinglow, 2016-06-28:
No, it is in `C:/Users/sock/robotology/opencv/include/opencv2`. I did manually put it in `build/opencv2` but that did not help either.

Tobias-Fischer, 2016-06-28:
Can you upload the (modified) `CMakeLists.txt` you are using? The uploaded version above is just the standard version without any modifications.

maxime-petit, 2016-06-28:
Oups yes sorry :D `C:/Users/sock/robotology/opencv/include/opencv2` is fine

Can you check also if you have another opencv directory installed? And check your variable environment (OpenCV_DIR and PATH).

I don't have opencv3 installed on my windows machines but I will take a look

sockchinglow, 2016-06-28:
In my `PATH` I have included the directory `C:\Users\sock\robotology\opencv\build\bin\Release`, the `OpenCV_DIR` is `C:\Users\sock\robotology\opencv\build`. The uploaded `CMakeLists.txt` differs from the standard version in the `include_directories` portion where I had added the `${OpenCV_INCLUDE_DIRS}` and `${OpenCV_LIBRARIES}`.

clement-moulin-frier, 2016-06-28:
Hi guys,

Same as @Tobias-Fischer , I am not very skilled in Windows compilation :/
As a light contribution however:
@max says that *"`C:/Users/sock/robotology/opencv/include/opencv2` is fine"* for `cmake` to find `opencv.hpp`. But from what I know the only path we provide to `cmake` is the `build` directory, so how could it find `include/opencv2`?

Tobias-Fischer, 2016-06-28:
Hmmm, sorry, I can't help in this case .. Out of ideas for remote Windows debugging ..

pattacini, 2016-06-28:
There shouldn't be any difference in this case between Windows and Linux, as we're using cmake just for that purpose :wink: It must be a problem due to how opencv has been installed and/or how `cvz` might be in sync with latest opencv releases.

I cannot go through the compilation of `cvz` given that I don't have `GTK2`, which is a required dependency. I've got also a concern about `cvz`: it's a project _per se_ which we don't use as far as I know and therefore I'd really suggest to hive it off from wysiwyd and provide it from a dedicated repository. Is there any plan to use it within wysiwyd?

I'm leaving to Leipzig and cannot help until next Tuesday.
However, I'm pretty confident you'll have sorted it out by then: it's just a matter of compiling code :smile: 

Getting shot of `GTK2` might also help...

Tobias-Fischer, 2016-06-28:
@sockchinglow By the way, what's the output of `git log -1`? As a quick fix, you can set `WYSIWYD_USE_GTK2` to `OFF`, it's only used by the `cvz` stuff.
@pattacini Happy to remove `cvz` as long as the guys at UPF are happy as well @jypuigbo @clement-moulin-frier - I agree it's a waste of time trying to get a module compiled we don't use.

pattacini, 2016-06-28:
@Tobias-Fischer I think `WYSIWYD_USE_GTK2` is an output variable, thus not used as input and therefore toggling it to `OFF` won't have any effect: it'll be rewritten to `ON`.

To disable `cvz` compilation @sockchinglow should comment out these lines:

1. https://github.com/robotology/wysiwyd/blob/master/main/src/libraries/CMakeLists.txt#L8
2. https://github.com/robotology/wysiwyd/blob/master/main/src/modules/CMakeLists.txt#L35

pattacini, 2016-06-28:
@sockchinglow can you compile just `iol2opc` to verify that the remaining modules are ok with your installation of `opencv-3.0.0`?

- It that's the case, then it's very likely that `cvz` must be upgraded.
- If `iol2opc` fails to compile, then your installation of `opencv-3.0.0` could be wrong somehow.

Tobias-Fischer, 2016-06-28:
@pattacini 
`WYSIWYD_HAS_GTK2` is the output variable, as far as I know `WYSIWYD_USE_GTK2` has the desired effect of being able to switch compilation off although the library is present.

pattacini, 2016-06-28:
@Tobias-Fischer you're right, although I've bad memories of playing with those cmake vars.

sockchinglow, 2016-06-29:
I managed to build the `wysiwyd` project after removing the `cvz` module! In addition to commenting the lines that @pattacini mentioned, I also "turned off" the variables `WYSIWYD_USE_GTK2` and, just in case, `ICUB_USE_GTK2`.

Thanks for your help @Tobias-Fischer @pattacini @maxime-petit @clement-moulin-frier! Since we don't use the `cvz` module, I guess even though we still don't know what the issue is we can consider this issue resolved.

closed 2016-06-29 by sockchinglow