robotology-legacy/wysiwyd #145: wrdac Swig (Python) bindings

Tobias-Fischer, 2016-07-15:
Hi,
@clement-moulin-frier suggested in https://github.com/robotology/yarp/issues/822 that it would be nice to have Python bindings for wrdac. I agree it would be a nice thing to have (seeing that both UPF and USFD work with Python); and it shouldn't be too hard to implement.

I'll assign @clement-moulin-frier for now.

Best, Tobi

clement-moulin-frier, 2016-07-15:
Thanks for opening it here @Tobias-Fischer 

Besides the fact of working with Python, it would also be a convenient way to interactively use some of our libraries, even for people that don't code modules in Python. For example we could use the `opcClient`  functionalities in an interactive Python shell like `ipython` in order to flexibly change the `OPC` state on the fly (instead on relying on the not-that-easy `rpc` syntax, or coding `populateSpecificX` functions in `opcPopulater` for each specific need).

I think the way to proceed is to adapt the [`CMakeLists.txt`](https://github.com/robotology/yarp/blob/master/bindings/CMakeLists.txt) provided for `yarp` bindings within the `wysiwyd` context. 
I am giving it a try. However I'm not completely comfortable with neither `Cmake` nor `Swig`, so any advice is of course welcome :)

I'll post here the progress and problems I encounter.

Cheers

Tobias-Fischer, 2016-07-15:
Hi,
It's probably easier to adapt one of the sample `cmake` scripts (http://www.swig.org/Doc1.3/Introduction.html#Introduction_build_system) rather than the one provided by yarp.
Let me know which issues you run into, I might be able to help. However I haven't worked with SWIG yet either.

Good luck! Tobi

Tobias-Fischer, 2016-08-17:
Did you get to work on this @clement-moulin-frier?

Tobias-Fischer, 2016-10-03:
Any news on this @clement-moulin-frier? Otherwise will close as `wontfix`.

closed 2016-10-22 by Tobias-Fischer

clement-moulin-frier, 2016-10-28:
Sorry I missed this one. I tried but no success. Certainly possible though if more time to allow on it.