robotology-legacy/wysiwyd #155: Adding calibration ability for subSystem_KARMA with iol

towardthesea, 2016-09-07:
This calibration ability should be similar to `subSystem_ARE` https://github.com/robotology/wysiwyd/blob/devel/main/src/libraries/wrdac/src/subsystems/subSystem_ARE.cpp#L42 
The problem is `KARMA` (in this current state) chooses the hand by itself after considering 2 solutions of 2 arms!

pattacini, 2016-09-08:
Hey @towardthesea @Tobias-Fischer 

According to [`karma` documentation](http://robotology.github.io/karma/doxygen/doc/html/group__karmaMotor.html), there's a simple way to force the use of one given arm, which resorts to the following _rpc service_:

```
[tool] [attach] arm x y z
```

where we would basically set `x y z` equal to `0.0 0.0 0.0` and `arm` equal to `left` or `right`.

In this manner, inside `subSystem_KARMA` level, one can conveniently choose which arm to move before initiating a `push` and/or `draw` action and thus request for the correct calibration hint.

I wouldn't expect very complicated actions, therefore the selection between `left` and `right` might be based on the `y` coordinate of the object to `push` / `draw`.

How do you see this?

pattacini, 2016-09-08:
#### Side note

I believe `subSystem_ARE` also forces the arm selection somehow through the `API` and or in a way hidden to the user.

Tobias-Fischer, 2016-09-08:
Hi,
This sounds good to me. I'll check with @towardthesea so he can start working on it.

Regarding `subSystem_ARE`: Yes, indeed this is the case.
```
    // always choose hand
    if (hand.empty())
    {
        if (handToUse.empty())
        {
            hand=(target[1]>0.0?"right":"left");
            options.addString(hand.c_str());
        }
        else
            hand=handToUse;
    }
```

Tobias-Fischer, 2016-09-09:
So two things need to be done:
1) Choosing the left/right hand for the actions
2) Applying the correction from `iolReachingCalibration`
@towardthesea is on it.

towardthesea, 2016-09-10:
Hi @Tobias-Fischer , @pattacini 

1. Choosing hand was implemented for `calibration before actions`. The chosen for actions are ongoing!
2. Correction from `iolReachingCalibration` was implemented and tested but there are still error (large in some cases) in the calibrated coordinates of recognized objects

Tobias-Fischer, 2016-09-10:
Hi @towardthesea , don't worry about the coordinates which are returned by `iolReachingCalibration`. They are wrong because the calibration we've done wasn't great. Just open a pull request as soon as choosing the hand also works for the KARMA action, and that's good :)!

towardthesea, 2016-09-12:
Hi @pattacini @Tobias-Fischer 
About the `table calibration`, I knew that there is such kind of `rpc` command to ask robot to check the table high and store the value as an object in `opc`. I meant in the email that should we automate the whole process and let the robot call the `table calibration` by itself (in case of we deal with table top scenarios) as initialized procedure, rather than we have to do it every time starting the robot. Then for sure KARMA has to get this `table high` to guarantee each action on the table.

Tobias-Fischer, 2016-09-12:
Hi @towardthesea,
I don't think we should do the calibration procedure every single time. There is no need for this. The table height typically doesn't change, nor does the kinematics of the robot (besides after broken cables etc.). So the calibration only needs to be done every couple of months, rather than at each start of the robot.
What do you think @pattacini?

Best, Tobi

pattacini, 2016-09-12:
@Tobias-Fischer is right. 

#### Table calibration
We need to run it only when the table itself changes or we're unsure the height is well found. This info is stored permanently within `OPC`.

To this end, it suffices to do once:
`$ echo "calib table" | yarp rpc /actionsRenderingEngine/cmd:io`

#### iolReachingCalibration
We need to run this process always after robot [fine calibration](http://wiki.icub.org/wiki/ArmFineCalibration) of course. Therefore, one-two months could be a reasonable timeline. However, it's a good rule to test the calibration outcome with a `touch` action.

towardthesea, 2016-09-12:
To sum up, the `table height` will be checked `manually` through `rpc` command then will be stored to `OPC`!

The next thing is that should the `table height` be read programmatically and/or through `ini` file? What are your opinion? 

If we go for the first option, I am not sure if we can use `OPC` class (`WYSIWYD` way) as we use for other `object`, because the `table` has only `table_height` parameter. As far as I checked, there is no such suitable `output` for it.  Please kindly give me suggestion!

Tobias-Fischer, 2016-09-12:
Hi @towardthesea,
I would add a `SubARE` within `SubSystem_KARMA` (similar to `SubABM` and `SubATT`). Then, use the `getTableHeight()` method to get the table height from `ARE`.
I am not sure whether the table height is really stored in the `OPC` or in a `ini` file.

Best, Tobi

pattacini, 2016-09-12:
`ARE` has two concurrent ways to store the `table height` information and retrieve it back at start-up:

1. In `OPC` by connecting `/actionsRenderingEngine/OPC:io` to `/OPC/rpc`. However, as you have already figured out, the format used by `ARE` here is not `wrdac` compatible. If `wrdac` will ignore data with unknown format we could go the whole hog, but I'm afraid it's likely not the case, since `wrdac` will get somehow disrupted 😞 
2. In the `kinematic_offsets.ini` file. I believe this option has lowest priority wrt comm with `OPC`.

Therefore, everything is basically in place with **option 2**.

Certainly, getting the `table height` by means of `SubARE` would be the proper approach. Nonetheless, consider that every time we instantiate a `SubARE` layer, a bunch of yarp ports are created. which eventually reduces the amount of resources we have.

Nonetheless, designing a more principled way is an overkill at this stage.

pattacini, 2016-09-16:
Can this be closed?

Tobias-Fischer, 2016-09-16:
I prefer to see the PR merged first :)

towardthesea, 2016-09-19:
I think we can close this issue also! 
Please kindly open it again if you find problem in the calibration with `KARMA subSystem`!

closed 2016-09-19 by towardthesea