robotology-legacy/wysiwyd #162: Object recognition problem with himrep

pattacini, 2016-10-02:
@dcam0050 is having the following recognition problem:

`himrep` keeps saying:
```
Trying to start Synchronization with OPC...
```
And once connected to `iol2opc` says this:
```
command received: [list]
Sending list request: objList
Received reply: ack ()
```
After which, `iol2opc` keeps sending:
```
[INFO]Sending classification request: [clas] ()
[INFO]Received reply: ()
```
to which `himrep` replies:
```
command received: [clas] ()
```

pattacini, 2016-10-02:
Hi @dcam0050,

`himrepClassifier` connects to `OPC` only within the `IOL` settings, not within WYSIWYD scenario.

In our case, instead, `himrepClassifier` functions as a _man-in-the-middle_ between `iol2opc` and `linearClassifierModule`, to which it forwards the requests received from `iol2opc` giving back the replies: see for example this [**code**](https://github.com/robotology/iol/blob/master/src/himrepClassifier/classifier.cpp#L514-L522) that handles the **`list`** request.

I suspect that `linearClassifierModule` starts off without finding the `IOL2OPCDatabase` folder in its context (or, equivalently, it finds an empty one). Can you check this?

Tobias-Fischer, 2016-10-03:
I might have an idea. It seems like `iol2opc` does not send any objects to `himrep` to classify. @dcam0050 told me that `lbpExtractor` does detect objects though; so I assume that `blobExtractor.read(false)` (L128 module.cpp) returns something (please confirm @dcam0050). 
Thus I reckon that the x/y-bounds are wrongly setup in the `config.ini`, which causes `skimBlobs` to not add the blobs to `skimmedBlobs`. Can you please report the values for `skim_blobs_x_bounds` and `skim_blobs_y_bounds`?
Tobi

Please replace `getBlobs` by the code below and report on the output.
```
Bottle IOL2OPCBridge::getBlobs()
{
    // grab resources
    mutexResources.lock();

    if (Bottle *pBlobs=blobExtractor.read(false))
    {
        yInfo("All blobs list: %s",pBlobs->toString().c_str());
        lastBlobsArrivalTime=Time::now();
        lastBlobs=skimBlobs(*pBlobs);
        yInfo("Received blobs list: %s",lastBlobs.toString().c_str());

        if (lastBlobs.size()==1)
        {
            if (lastBlobs.get(0).asVocab()==Vocab::encode("empty"))
                lastBlobs.clear();
        }        
    }
    else if (Time::now()-lastBlobsArrivalTime>blobs_detection_timeout)
        lastBlobs.clear();

    // release resources
    mutexResources.unlock();

    return lastBlobs;
}
```

dcam0050, 2016-10-03:
I found out yesterday that there was another problem that was causing iol2opc to crash but stay open. When sending a `[clas]` command, not all return paths made sure there was something present in the bottle that was being sent as a reply and as a consequence iol2opc would report a yarp bottle read failure and get stuck there.

@Tobias-Fischer I will be trying the code out now. Because iol2opc places a green box around the detected blob but it is very intermittent so nothing is showing up on the histogram window.

@pattacini This is the code I changed should you want to check it out . Areas I changed have a `//here` written. Also I found a lot of curly brackets surrounding if statements missing. Not sure if that is on purpose or not.

```
void Classifier::classify(Bottle *blobs, Bottle &reply)
{
    if (blobs==NULL)
    {
        printf("Null Blobs \n");
        reply.addList();
        return;
    }

    if (blobs->size()==0)
    {
        printf("Blob size 0 \n");
        reply.addList();
        return;
    }

    if ((imgInput.getInputCount()==0)     || (imgSIFTInput.getInputCount()==0) ||
        (featureInput.getInputCount()==0) || (scoresInput.getInputCount()==0)  ||
        (rpcClassifier.getOutputCount()==0))
    {
        printf("Not enough image connections \n");
        reply.addList();
        return;
    }

    // read object classes
    Bottle cmdObjClass,objClassList;
    cmdObjClass.addString("objList");
    rpcClassifier.write(cmdObjClass,objClassList);
    if (objClassList.get(0).asString()!="ack")
    {
        //here
        printf("objClassList (0) not ack \n");
        reply.addList();
        return;
    }

    Bottle *objList=objClassList.get(1).asList();
    if (objList==NULL)
    {
        //here
        printf("objClassList (1) empty list \n");
        reply.addList();
        return;
    }

    if (objList->size()==0)
    {
        printf("objClassList (1) not empty list \n");
        for (int b=0; b<blobs->size(); b++)
        {
            Bottle &blob_scorelist=reply.addList();
            blob_scorelist.addString(blobs->get(b).asList()->get(0).asString().c_str());
            blob_scorelist.addList();
        }

        return;
    }

    // start recognition mode
    Bottle cmdClass,classReply;
    cmdClass.addString("recognize");
    rpcClassifier.write(cmdClass,classReply);
    if (classReply.get(0).asString()=="nack")
    {
        printf("linear classifier recognize failure\n");
        reply.addList();
        return;
    }

    // read image
    ImageOf<PixelRgb> *image=imgInput.read();
    if (image==NULL)
        //here
        printf("null input image \n");
        reply.addList();
        return;

    IplImage *imgC=(IplImage*)image->getIplImage();

    // classify each blob
    printf("Start classification\n");
    for (int b=0; b<blobs->size(); b++)
    {
        // list of the scores
        Bottle &blob_scorelist=reply.addList();
        // name of the blob
        blob_scorelist.addString(blobs->get(b).asList()->get(0).asString().c_str());
        // list of scores
        Bottle &scores=blob_scorelist.addList();
        // retrieve bounding box
        Bottle *bb=blobs->get(b).asList()->get(1).asList();
        int x_min=(int)bb->get(0).asDouble();
        int y_min=(int)bb->get(1).asDouble();
        int x_max=(int)bb->get(2).asDouble();
        int y_max=(int)bb->get(3).asDouble();

        if (x_min>5)
           x_min-=5;

        if (y_min>5)
           y_min-=5;

        if ((x_max+5)<imgC->width)
           x_max+=5;

        if ((y_max+5)<imgC->height)
           y_max+=5;

        // crop image
        ImageOf<PixelRgb> croppedImg;
        croppedImg.resize(x_max-x_min,y_max-y_min);
        cvSetImageROI(imgC,cvRect(x_min,y_min,x_max-x_min,y_max-y_min));
        cvCopy(imgC,(IplImage*)croppedImg.getIplImage());
        cvResetImageROI(imgC);

        // send image to SC
        printf("sending cropped image to caffeCoder \n");
        imgOutput.write(croppedImg);

        // read coded features
        Bottle fea;
        featureInput.read(fea);
        ImageOf<PixelRgb> *imgSift=imgSIFTInput.read();
        if ((fea.size()==0) || (imgSift==NULL))
            printf("imgSift return from caffe = null \n");
            return;

        x_max=std::min(x_min+imgSift->width(),image->width()-1);
        y_max=std::min(y_min+imgSift->height(),image->height()-1);
        IplImage *iplSift=(IplImage*)imgSift->getIplImage();

        cvSetImageROI(iplSift,cvRect(0,0,x_max-x_min,y_max-y_min));
        cvSetImageROI(imgC,cvRect(x_min,y_min,x_max-x_min,y_max-y_min));
        cvCopy(iplSift,imgC);
        cvResetImageROI(imgC);

        // send feature to classifier
        printf("send feature to linearClassifier \n");
        featureOutput.write(fea);

        // read scores
        Bottle class_scores;
        scoresInput.read(class_scores);
        if (class_scores.size()==0)
            printf("size of received scores = 0 \n");
            return;

        // fill the list of the i-th blob
        printf("Scores received: ");
        for (int i=0; i<objList->size(); i++)
        {
            Bottle *obj=class_scores.get(i).asList();
            if (obj->get(0).asString()=="background")
               continue;

            Bottle &currObj_score=scores.addList();
            currObj_score.addString(obj->get(0).asString().c_str());
            double normalizedVal=((obj->get(1).asDouble())+1.0)/2.0;
            currObj_score.addDouble(normalizedVal);
            printf("(%s %g) ",obj->get(0).asString().c_str(),normalizedVal);
        }

        printf("\n");
    }

    if (imgSIFTOutput.getOutputCount()>0)
    {
        imgSIFTOutput.prepare()=*image;
        imgSIFTOutput.write();
    }
}```

dcam0050, 2016-10-03:
@pattacini Yes linearClassifierModule starts out with a missing IOL2OPC directory

dcam0050, 2016-10-03:
@Tobias-Fischer The values I am using for `skim_blobs_x_bounds` and `skim_blobs_y_bounds` are the default values that come with the iol2opc `config.ini` 

```skim_blobs_x_bounds         (-0.70 -0.10)
skim_blobs_y_bounds         (-0.50  0.50)```

dcam0050, 2016-10-03:
@Tobias-Fischer This is part of the log after I did the change you proposed:

When a blob is actually received the parameters of the blob are sent to `himrep` with a `[clas]` command. But most of the `[clas]` commands being sent are still empty. 

Does this affect the recognition of the object?

```[INFO]Received reply: ()
[INFO]Sending classification request: [clas] ()
[INFO]Received reply: ()
[INFO]Sending classification request: [clas] ()
[INFO]Received reply: ()
[INFO]Sending classification request: [clas] ()
[INFO]Received reply: ()
[INFO]Sending classification request: [clas] ()
[INFO]Received reply: ()
[INFO]Sending classification request: [clas] ()
[INFO]Received reply: ()
[INFO]Sending classification request: [clas] ()
[INFO]Received reply: ()
[INFO]Sending classification request: [clas] ()
[INFO]Received reply: ()
[INFO]All blobs list: (151.0 120.0 202.0 189.0)
[WARNING]get3DPosition failed
[INFO]Received blobs list: 
[INFO]Sending classification request: [clas] ()
[INFO]Received reply: ()
[INFO]Sending classification request: [clas] ()
[INFO]Received reply: ()
[INFO]Sending classification request: [clas] ()
[INFO]Received reply: ()
[INFO]Sending classification request: [clas] ()
[INFO]Received reply: ()
[INFO]Sending classification request: [clas] ()
[INFO]Received reply: ()
[INFO]Sending classification request: [clas] ()
[INFO]Received reply: ()
[INFO]Sending classification request: [clas] ()
[INFO]Received reply: ()
[INFO]Sending classification request: [clas] ()
[INFO]Received reply: ()
[INFO]Sending classification request: [clas] ()
[INFO]Received reply: ()
[INFO]Sending classification request: [clas] ()
[INFO]Received reply: ()
[INFO]Sending classification request: [clas] ()
[INFO]Received reply: ()
[INFO]All blobs list: (151.0 120.0 202.0 189.0)
[WARNING]get3DPosition failed
[INFO]Received blobs list: 
[INFO]Sending classification request: [clas] ()
[INFO]Received reply: ()
[INFO]Sending classification request: [clas] ()
[INFO]Received reply: ()
[INFO]Sending classification request: [clas] ()
[INFO]Received reply: ()
[INFO]Sending classification request: [clas] ()
[INFO]Received reply: ()
[INFO]Sending classification request: [clas] ()
[INFO]Received reply: ()
[INFO]Sending classification request: [clas] ()
[INFO]Received reply: ()
[INFO]Sending classification request: [clas] ()
[INFO]Received reply: ()
[INFO]Sending classification request: [clas] ()
[INFO]Received reply: ()
[INFO]Sending classification request: [clas] ()
[INFO]Received reply: ()
[INFO]Sending classification request: [clas] ()
[INFO]Received reply: ()
[INFO]Sending classification request: [clas] ()
[... 148 lines cut ...]
[INFO]Received reply: ()
[INFO]Sending classification request: [clas] ()
[INFO]Received reply: ()
[INFO]Sending classification request: [clas] ()
[INFO]Received reply: ()
[INFO]Sending classification request: [clas] ()
[INFO]Received reply: ()
[INFO]Sending classification request: [clas] ()
[INFO]Received reply: ()
[INFO]Sending classification request: [clas] ()
[INFO]Received reply: ()
[INFO]Sending classification request: [clas] ()
[INFO]Received reply: ()
[INFO]Sending classification request: [clas] ()
[INFO]Received reply: ()
[INFO]Sending classification request: [clas] ()
[INFO]Received reply: ()
[INFO]Sending classification request: [clas] ()
[INFO]Received reply: ()
[INFO]Sending classification request: [clas] ()
[INFO]Received reply: ()
[INFO]Sending classification request: [clas] ()
[INFO]Received reply: ()
[INFO]Sending classification request: [clas] ()
```

Tobias-Fischer, 2016-10-03:
Well ` [WARNING]get3DPosition failed` gives a pretty good hint. Are you sure SFM is working fine? Can you try sending `Rect 147 117 7 7 2` as well as `Point 150 120` to `/SFM/rpc` and report the replies?

dcam0050, 2016-10-03:
Yes I currently have a big problem with `SFM`. I thought it was OK because when I tested the disparity map, I did it with the iCub looking head on which looked really good but as soon as I `home all` with `ARE` almost the whole disparity map turns black. So now I'm trying to recalibrate but no success. I've had 2 questions come up:

When you perform `stereoCalib`, should the eye vergence be at 5deg or at 0? (It was at 0 when I did mine)

Should `SFM` be run from the same computer that runs `camCalib` because it seems like `SFM` reads `icubEyes.ini` from `stereoCalibration` context

Tobias-Fischer, 2016-10-03:
1) I think at 5deg, but @pattacini please confirm
2) Yes! I think this is nowhere documented, but I'm 99% sure that `camCalib` and `SFM` should run on the same machine. @pattacini shall we document this somewhere?

pattacini, 2016-10-03:
1) It depends. The only important point is that you launch `ARE` with the same vergence used for calibration. We usually use `5 deg`.

2) `camCalib` and `SFM` can run everywhere. We in fact launch them on separate machines.

pattacini, 2016-10-03:
I think @Tobias-Fischer is definitely right: there must be a stereo problem with your robot which prevents from inserting objects in the database.

@dcam0050 could you please open a PR. I won't go through the code you posted for obvious reasons :smile:

dcam0050, 2016-10-03:
@pattacini are the separate machines you mentioned connected via the same .local filesystem though? Because I'm running `SFM` on the cuda machine and `camCalib` on the laptop and they don't share the .local filesystem. Regarding himrep I will open a PR.

pattacini, 2016-10-03:
Well, `SFM` needs to access intrinsic camera parameters provided by `camCalib` during calibration, this is somehow understood: it's even not the only module requiring that file (e.g. `iKinGazeCtrl` needs it too). Along the same line, if you want to launch WYSIWYD suite part on the laptop and part on the cluster, again, they have to retrieve the same information.

My considerations above were beyond the managing of the cluster.

Therefore, you ought to make sure that all the machines can share the same info.

Tobias-Fischer, 2016-10-03:
Hi @pattacini, it might be well understood for you guys, but for people new to the iCub this fact might not be too obvious ;-).

pattacini, 2016-10-03:
I can see your point @Tobias-Fischer but this is a general approach to the eco-system, it's not related to `SFM` + `camCalib` only, hence also suggesting to run both on the same machine is not totally correct in my opinion.

pattacini, 2016-10-03:
What I wanted to pointed out is that if you connect a laptop to a properly configured cluster (meaning its machines share the `.local\share\yarp` thing plus other stuff), then you really need to be careful because you could come across downsides like the one above.

I gave this for granted, but it could be worth putting this info somewhere in the WYSIWYD instructions.

dcam0050, 2016-10-03:
I agree with @Tobias-Fischer here because I'm just finding this out after a year and a half ... might actually explain a lot of the weird behaviour that we see here in Sheffield

dcam0050, 2016-10-03:
Ok so after configuring all computers to have the same shared .local files, SFM calibrated immediately and it's a great disparity map ... also I had previously noted that lbpExtract and iol2opc seemed to operate independently with iol2opc sending an empty `[clas]` command in between receiving blob information. That is also gone. The issue now is that after an amount of `[clas] (blob_x(a b c d))`, caffeCoder receives an image and carries out a classification. 

The problem is that upon finishing the classification, the chosen blob stops appearing in `/viewer/iol/histogram` and no scores are displayed. All vision modules continue operating as usual, none of them crash but I have noticed an accumulation of `unknown_object_x` in `IOL2OPCDatabase` until eventually `linearClassifierModule` gives the following error and crashes.

```
WARNING: actred and prered too small
nClass: 2 nPositive: 0 nNegative: 16
Floating point exception (core dumped)
```

Right now I'm still using `caffeCoder` with `caffe-rc3` compiled using `opencv 2.4.8` ... maybe that is the issue. I'll try compiling `caffe` master with `opencv 3.1` and subsequently `himrepClassifier` and `linearClassifierModule` with `opencv3.1` and see how it turns out.

dcam0050, 2016-10-04:
So yesterday I did not manage to compile caffe master with opencv 3.1 so I compiled both `caffe` and `caffeCoder` with opencv 2.4.12 and it compiles but when using it together with iol2opc I get strange behaviour. `/viewer/iol/histogram` stops displaying the blob it is currently classifying when `caffeCoder` has finished processing an input image. Also the `IOL2OPCDatabase` is filling up with a new `unknown_object_x` every time `linearClassifierModule` tries to `[train]` the SVM. Then eventually `linearClassifierModule` crashes with the following error:

```WARNING: actred and prered too small
nClass: 2 nPositive: 0 nNegative: 6
Floating point exception (core dumped)```

Tobias-Fischer, 2016-10-04:
Did you try deleting the `IOL2OPCDatabase` folder? Also, can you record a video with the different module outputs visible so it's easier to understand for us what's going on?

pattacini, 2016-10-04:
I was thinking to arrange for a remote chat session, but I'm afraid I'll be available only in two days during IFI...

dcam0050, 2016-10-05:
@Tobias-Fischer I'll be recording a video today

@pattacini What are the dates you are available in? My timetable is pretty flexible right now

pattacini, 2016-10-05:
Don't know yet with the necessary precision. Perhaps tomorrow.

Btw, may I ask you what catastrophic event happened to your system? 😄 
I remember object recognition was perfectly working in Sheffield.

dcam0050, 2016-10-06:
@ugo Updates happened haha

Ok so I figured out what went wrong. 

1) Initially iol2opc was freezing up which led me to modify the himrepClassifier to return something with every return which resulted in iol2opc not freezing up so I thought it was progress.

2) Then I learned about the requirement for a common .local directory which fixed SFM and seemed to sync blobs arriving from lbpExtract with [clas] requests from iol2opc to himrepClassifier but now it was crashing when a classification was made by caffeCoder

3) Reverted back the himrepClassifier code to its original state and everything worked.

So in summary, not having a shared .local/share/yarp folder was the cause of all this trouble! Thanks for helping me out!

dcam0050, 2016-10-06:
Previously it was working 'fine' in Sheffield even though I did not have a shared .local folder because I had initially configured the system to share both .local and /usr/local/src/robot but I then removed the linking on both so I had .local files which were a copy of when they were configured together

pattacini, 2016-10-06:
Ok... happy to know that this big problem just deflated. I was started to be scared about that 😄

closed 2016-10-06 by pattacini