robotology-legacy/wysiwyd #163: Adding a waypoint for homing after push actions

jypuigbo, 2016-10-05:
It would be safer to have an intermediate waypoint for going to `home` position after a push action. 
Maybe @towardthesea could help on that?

The changes for this behavior are being made in the branch `feature/upf/pushBehaviors`.

towardthesea, 2016-10-07:
Hi @jypuigbo, @sockchinglow  
A new method was added to `subSystem_KARMA` to return robot arm after any push actions (but before home call). It was tested in IIT without any problem.
Could you please check it with `behaviorManager`! Then we can make a PR!

Cheers,
Phuong

Tobias-Fischer, 2016-10-27:
Was this tested in the meantime? Can we merge?

towardthesea, 2016-10-28:
@Tobias-Fischer I tested in IIT but still waiting the full test of UPF with behaviorManager. Should we wait?

clement-moulin-frier, 2016-11-10:
Hi,

Sorry guys we have been delayed a lot these last weeks. The rush is ending tomorrow, so Monday we test it sure (our iCub is working fine now).

Clément

pattacini, 2016-11-10:
Hi @clement-moulin-frier,

Then, do you have any news on https://github.com/robotology/icub-support/issues/321?

jypuigbo, 2016-11-14:
It is solved!

Tobias-Fischer, 2016-11-15:
Does that mean you tested this and we can close the issue @jypuigbo? 
To be honest, I am still not sure whether this issue exists just to test something which is already in `devel`, or test something in another branch which still needs to be merged into `devel`. Could you please let me know @towardthesea?

clement-moulin-frier, 2016-11-15:
Hi,

We are testing the push behaviors now.
@towardthesea : to do so, do you confirm we have to run the modules in `main/src/examples/KARMAiCubClientExample/KARMAiCubClientExample.xml`?
from the branch `feature/upf/pushBehaviors`

sockchinglow, 2016-11-15:
@towardthesea we did not see any further problems with this during the test. Thanks!

closed 2016-11-15 by sockchinglow