robotology-legacy/wysiwyd #166: Perspective taking demonstration

Tobias-Fischer, 2016-11-07:
After the integration of the perspective taking module at the last integration meeting in Barcelona, we must now decide how to demonstrate this module in the common demo.
@maxime-petit and I spoke about it, and the neatest way seems to be increasing the saliency of an object which is hidden to the human. Then, when expressing knowledge, the robot prefers talking about this object, and can say "Do you know that there is a banana, but it is behind the wall so you cannot see it". Another option would be to grasp the object and drop it somewhere where it is visible.

Tobias-Fischer, 2016-12-05:
Update on this: We thought a nice way of demonstrating the perspective taking knowledge is estimating the reachability of the human. This will be done by using the forward models learned by @martz21 on the robot, then finding the kinematic structure correspondences by @hjchang, and finally doing the mental rotation using my perspective taking module. We'll do this in a probabilistic manner, where we can encode how hard it is for the human to reach an area.
The plan is to demonstrate this at the beginning of the table top scenario using the knowledge expression behavior.

Did I forget something @maxime-petit?

maxime-petit, 2016-12-05:
Yop!

Just as a small details: the triggering of the demonstration before the demo might actually not be from knowledge expression (even if we can potentially trigger it like this) in order to "be in control". Unless it is the only knowledge the robot knows (and pre-setting the drive to have knowledge expression first) it is not guarantee to be triggered.

./Max

sockchinglow, 2016-12-05:
For the Y3 tabletop scenario, the current definitions for reachable zones of the human and the iCub are somewhat arbitrarily set. If we are able to integrate this perspective taking into the `sensationManager`, it is possible to define the reachable area of the human based on the real-time position of the human partner instead. This could be shown by the human moving around the table (to a certain extent, of course), and the iCub showing that in the appropriate situation it asks the human to reach for the object and in others it declares that the object is not in either of their reachable areas.

Tobias-Fischer, 2016-12-05:
@sockchinglow Yes that would be the ideal :+1:

Tobias-Fischer, 2016-12-15:
Update on this: there is a PR which supports getting the reachable area. For now, it is the bounds defined in `iol2opc`. I am afraid that any self-learned reachable area will be too fragile, as the bounds between the different areas are very, very small due to the constraints given by the small area of the robot in terms of pulling on the one side, and perception on the other side. Therefore, I suggest we use the script only for visualizing the areas, but not for actual computations.
/cc @maxime-petit @hjchang

Tobias-Fischer, 2017-02-09:
Update: We definitely won't use the module for detecting the reachable area during the demo. However, we try to show the learning process of the reachable area in a video / image.

Tobias-Fischer, 2017-02-13:
We got the results, I'll check with Yiannis whether to show this. Will close here, last issue .. whoop whoop.

closed 2017-02-13 by Tobias-Fischer