robotology-legacy/wysiwyd #168: KARMA application in WYSIWYD demo

sockchinglow, 2016-11-15:
We have managed to test the `KARMA` behaviours and have found the following problems that need to be resolved before it can be run as part of the demo:

- Both `PushingRight` and `PushingLeft` creates a scooping behaviour rather than a horizontal movement. This might be because the wrist joint angle is fixed and the only way to move from the start to end point is to execute that movement. Needless to say, it would be dangerous when done above a table.

- In the `KARMAexample`, when pushing forward the hand is ~20cm above the table. When pulling, the height is correct but the wrist/hand position is not angled sufficiently to pull any object so it just caresses it.

- Most of the movements are rather quick; could we modify the speed either through an `RPC` command or a variable in the config file?

Tobias-Fischer, 2016-11-15:
Can we try summarizing all `KARMA` related issues within one issue? So far, we have four: this one, https://github.com/robotology/wysiwyd/issues/167, https://github.com/robotology/wysiwyd/issues/165 and https://github.com/robotology/wysiwyd/issues/163

sockchinglow, 2016-11-15:
@Tobias-Fischer the issues are not directly related except for #163 (which we have just resolved). The others are more on how to tune `KARMA` to fit the demo.

towardthesea, 2016-11-16:
Hi @sockchinglow , thanks for updates. I have some questions to make your points clearer:
  1. Have you tried to run `subSystem_KARMA` with newest update of the branch (I made a `push` on Friday afternoon)?
  2. Have you checked the `table height` and does the `height` make sense?
  3. Can you explain your meaning of `scooping behavior`: It is better to understand with short videos (or similar) of the behaviors. During my test with `KARMAexample` I couldn't recognize that behavior.
  4. If the `table height` is sufficient good, the distance between the table and the hand as ~20cm doesn't make sense to me, because I ran the same `example` many times and it didn't happen as you described.
  5. The `pulling` behavior with a right `hand angle` is taken care by a new version of `karma` (https://github.com/towardthesea/karmaWYSIWYD) with the spirit that no any changes are required for higher level of WYSIWYD framework. The transition to `karmaWYSIWYD` will be painless (I promise). I will update with the link for the manual soon.
  6. The `motion speed` is also solved as in 5.

sockchinglow, 2016-11-16:
Hi @towardthesea, thanks for the response.

1. The files were built from what was on github yesterday, so I guess it's the newest version of `subSystem_KARMA`.
2. The `table height` was correct. The other actions were working fine in terms of height.
3. When moving from the starting point, the height of the hand moves downwards and then back upwards to the specified height when it reaches the ending point. The whole hand below the table at one point, but we did not measure it since the hand was moving.
4. We know it doesn't make sense, but that is how the iCub acts...it consistently raises its hand too high for only the forward pushing behaviour within the `KARMAexample`.
5. Noted, we will wait for the update regarding the `hand angle` and `motion speed`.

towardthesea, 2016-11-16:
As I promised, the new issue https://github.com/robotology/wysiwyd/issues/170 was created to introduce the new version of `karma`, named `karmaWYSIWYD`.

towardthesea, 2016-11-17:
Hi @sockchinglow , to be clear I will firstly describe what will happen when users call `push` action (including `left`, `right` and `front`) and `pull` action from `karma`:

**push** from `x0` to `x1`
  - move to `x0'` higher than `x0` an offset (currently set at 0.1m)
  - move down to `x0`
  - move to `x1`
  - come back to `x0`
  - move up 0.05m (new update provided by `subSystem_KARMA` from `UPF` request as this branch https://github.com/robotology/wysiwyd/issues/163)

**pull** from `x0` to `x1`
  - move to `x0'` higher than `x0` an offset (currently set at 0.05m)
  - move down to `x0`
  - move to `x1`
  - move up 0.05m (new update provided by `subSystem_KARMA` from `UPF` request as this branch https://github.com/robotology/wysiwyd/issues/163)

When users call `pushKarmLeft`, `pushKarmRight` and `pushKarmFront` the z-value in `x0` or `x1` above-mentioned will be replace by the `table_height` + offset (as shown here https://github.com/robotology/wysiwyd/blob/feature/upf/pushBehaviors/main/src/libraries/wrdac/src/subsystems/subSystem_KARMA.cpp#L343).

Hope that the insight of `karma` can provide a clearer idea of what is going on.

So the easiest way to check the behavior of your robot now is: 
  - Start new `karmaMotor` in a separate terminal
  - Re-run the `KARMAexample` without the table 
  - Provide us the log from the terminal of `karmaMotor` and `KARMAexample`

Waiting for your update.

Cheers,
Phuong

Tobias-Fischer, 2016-11-23:
Can we close this (as soon as #174 is merged)? @sockchinglow?

towardthesea, 2016-11-23:
#174 is merged, this should be closed then!

towardthesea, 2016-11-23:
Closed this by mistake! We should wait for testing and confirmation from UPF!
@sockchinglow please keep update asap!

sockchinglow, 2016-11-23:
@towardthesea We have just resolved another problem that we had, so we will test `karma` tomorrow. Naturally, we will keep you updated and hopefully this issue will be closed soon.

towardthesea, 2016-11-24:
@sockchinglow this video (https://www.youtube.com/watch?v=eyz9d6KdalY&feature=youtu.be) shows how the `KARMA example` run in our iCub. Does it similar at UPF?
Btw, new `pulling` methods are added (the `KARMA example` is changed also), you guys should update the `behaviorManager`.

jypuigbo, 2016-11-24:
Hi,

we have tested the new karmaWysiwyd module.

It works better than the previous one (more precise and safe) but there are still a few issues.

- This is what we get with `KARMAexample`. 
https://goo.gl/photos/8Tt3CqvvDZ21YxaY8
The main problem is for push and pull, it is quite different from the video posted by @towardthesea above. Pulling in particular is very weird. 

Then, running it from the BehaviorManager:
- Pushing in front works fine. However it was often missing the object due to errors in location from IOL. 
- We still have to implement the pulling behavior.

At this point it would be useful to have feedback from other labs also testing it, since it seems that we obtain quite different results between IIT and UPF. Could anyone try as well? This can be done by:
- Installing the karma fork as explained by @towardthesea  above
- switching to the branch `feature/upf/pushBehavior`
- running `karmaMotor`
- testing both:
    - `karmaExample` provided by @towardthesea 
    - the push behavior from `BehaviorManager`
        Modules:
        - Y2 demo @ inserm without the reactive layer (essentially `IOL` + dependencies + `icubGui` for feedback
        - Modify `default.ini` of behaviourmanager to include JUST the `pushingFront` behavior and use_ears set to false
        - Run `behaviorManager`
        - Send `pushingFront (object_name object_type)` to `/BejhaviorManager/trigger:i`

pattacini, 2016-11-24:
@jypuigbo just a quick side note

`feature/upf/pushBehaviors` doesn't seem to contain latest commits on `karma` since it is not rebased on top of `devel`.

pattacini, 2016-11-24:
@towardthesea can you also try out the instructions posted by @jypuigbo?
Or did you already do that?

Tobias-Fischer, 2016-11-25:
@maxime-petit and I will give the `karmaExample` a try today. As soon as `feature/upf/pushBehavior` is rebased on top of `devel`, we can give it a try as well.

towardthesea, 2016-11-25:
I am checking it now @pattacini @jypuigbo !

clement-moulin-frier, 2016-11-25:
@sockchinglow :  What's the status of rebasing `pushBehaviors` on top of `devel`? Let me know if you need help doing it.

sockchinglow, 2016-11-25:
@clement-moulin-frier @towardthesea @Tobias-Fischer : Due to the internal workshop, we would probably be unable to `rebase` and test the branch by the end of today. Nonetheless, we will work on that as soon as possible.

Tobias-Fischer, 2016-11-27:
@maxime-petit let's try this tomorrow. Our robot was on strike on Friday .. it didn't wanna start due to our random torso problems. Let's hope he'll behave tomorrow.

jypuigbo, 2016-11-28:
Update from upf after testing again post-rebaseing `devel`:
- `PullingBack` has been added as a behavior and the code in all behaviors has been cleaned in all others. 
- `PullingBack` and `PushingFront` behaviors now work quite well. 
- `PushingRight/Left` try to cross the table when moving (see video below)
- All: They seem to be using the wrist to push/pull the objects, what makes it more susceptible to miss-calibrations in stereovision (see video below)
- Finally, we observed that, if something goes wrong with one of the arms, the robot will try to push with the other wherever the object is, what makes dangerous movements that cross all over the table. 

[Table crossing](https://goo.gl/photos/mza3SGk2YApUUL9J7)
[Wrist pushing
]( https://goo.gl/photos/Riy444XdmUxwi78T8)

towardthesea, 2016-11-28:
@jypuigbo Could you declare the condition of the two videos? I cannot analyse without knowing the scenario. 

> Finally, we observed that, if something goes wrong with one of the arms, the robot will try to push with the other wherever the object is, what makes dangerous movements that cross all over the table.

Your above conclusion is not clear for me :)

sockchinglow, 2016-11-29:
@towardthesea `Table crossing` is when the iCub uses the right hand to push an object on its left, instead of using the left hand. `Wrist pushing` shows how the iCub uses the position of its wrist as the end effector instead of its palm/fingers to push objects. In other words, instead of using its palm to push things, it uses its wrist. This means that the "correct position" is offset by the distance from the wrist to the middle of the palm.

The conclusion was another way of describing what is in the video `table crossing`. So if it wants to push an object far on the left with the right hand, it reaches across the table even if it pushes the limits of its joints. Also, when returning to `home` position, sometimes parts of its arm goes below table height.

Hope that helps.

towardthesea, 2016-11-29:
Hi @sockchinglow @jypuigbo ,

By default, `karma` will choose the arm to conduct the action by itself if user doesn't define any arm to use. In particular, if the object on the `left side` wrt the x-axis, robot will use the `left arm` to push the object, then depending on the action, i.e `pushingLeft` or `pushingRight`, the `table crossing` happens. For this case, it is for `pushingRight`. So, if user doesn't want the `table crossing` happens, she/he can ask robot to `pushingRight` with `right arm` and `pushingLeft` with `left arm` but please consider if the object is too far to the left or the right side. From the low-level point of view, an designer cannot cover all situation but the API allows user to do it. So feel free to play :smile: 

All `push` actions should be done with the `palm` or the `back of hand`, rather than the `wrist`. For the `pull`, robot will form its hand as a rake and use the tip of finger to pull object. So the `wrist pushing/pulling` makes no sense to me. 

For `home`, it comes from `ARE`, basically will move the arm in a `straight line` to the `default home position`. If you call home after any `karma` actions, it should be safe and fine. Out of that context, I am not sure.

If you want, please provide us the conditions of experiment when the behaviors happened, i.e branch you are working on, commands, etc. then we can help to re-run the experiment and consider the bugs (if there is :smiley: )

pattacini, 2016-11-29:
I've got a rough idea of what might be happening that could potentially explain the so-called "_table crossing_". It has something to do with the distance between the starting and ending point of the push action. If that's too long, then the swing might end up being too curved.

I'd kind of have a workaround for this, but since it's not of an easy implementation, we should consider the "dimensions" of the push actions we deal with.

I feel to support @towardthesea's request to try to reproduce the exact experiment also here at IIT.

Are these [instructions](https://github.com/robotology/wysiwyd/issues/168#issuecomment-262834836) from @jypuigbo still valid and sufficient or do we need more?

jypuigbo, 2016-11-29:
@towardthesea Yes, I understand that not everything can/must be considered from the design point of view and much will be done from the user point of view. From my side, I was just stating what were the issues I was finding, to check that they made sense and we were not doing anything wrong from our side. In this regard, we just tested the predefined behaviors with the predefined positions while today we are planning on testing it all together with the positions we foresee for the final setup, to see the range of positions available. Still, the second comment, stating that push front and pull back behaviors work well enough (and actually, push right/left as well if we keep displacements small) still holds :) . 

For the wrist pulling, what do you think of the video then? It is a top view of trying to push the virtual object you see at the robot's right. The center of the object is aligned with the wrist, personally speaking. If you can, try the`opcPopulater` option that I suggest in [here](https://github.com/robotology/wysiwyd/issues/168#issuecomment-262834836) ( @pattacini instructions updated ) and see in the `icubGui` the position of the hand with respect to the object. Maybe it's a parallax error on our part. In any case, the real issue comes when we test it with `IOL` running, because then the position of the object is not 100% correct and it fails in the real world (in `icubGui`, the pulling/pushing would work fine). 
Also, we are re-calibrating `SFM` again to try to get the best possible accuracy. 

Finally (Finally!) what the last comment was referring to ( @towardthesea ) is specifically: 
- We had a conflict address with the left arm and `karmaMotor` (due to running the example and the `behaviorManager` at the same time, what should not happen normally)
- We run push behaviors over objects on the left
- The hand selected was the right hand and the behavior was run. 
- This implies that the right hand was dangerously going to the far left side of the table and was pushing an object. 
Again, this can be avoided by just not having objects too far, but still I thought it might be useful for you to know that this is happening, in case you want to make it more robust. For the demo, it is probably not a relevant issue. 

All the changes that I've made are pushed in the branch `feature/upf/pushBehaviors`, for any test.

jypuigbo, 2016-11-29:
Just in case it was not clear, to reproduce the wrist-pushing issue these are the exact steps: 
- Installing the karma fork as explained by @towardthesea  above
- switching to the branch `feature/upf/pushBehavior`
- running `karmaMotor`
- run `opcPopulater` and send `populateSpecific3` to `/opcPopulater/rpc` (`IOL` can be used instead, but the issue is not clearly seen, then)
- `OPC` and `ABM` (can be removed from `client.ini` for this test) need to be running
- running `icubGui` for feedback
- Modify `default.ini` of `behaviorManager` to include JUST the `pushingFront` behavior and `use_ears` set to `false` (see below)
- Modify `client.ini` in `behaviorManager` to have only `(ARE KARMA)` in `subsystems`
- Run `behaviorManager`
- Send `pushingFront (object unknown_1)` (this is the precise command, if you followed the instructions above, to run the same you see in the video) to `/BehaviorManager/trigger:i`

It would be useful, indeed if you can try it. 

********* top of our `default.ini` *********
```
use_ears        false

[BEHAVIORS]

behaviors       (pushingFront)

...
```

Tobias-Fischer, 2016-11-30:
Hi guys, @maxime-petit and I finally got around trying this. We have only tried the `KARMAiCubClientExample` however, and not the integration with `behaviorManager` (we can do this after @pattacini and @towardthesea tried this and if needed).
It works nicely on our robot. I guess the next step for @towardthesea and @pattacini is to try whether it actually works with objects as perceived by `iol2opc`.
Here is the video: https://drive.google.com/open?id=0BwyaFLNKyQcqSTdWenZLSjRIbHM

Best,
Tobi

towardthesea, 2016-12-01:
Hi guys,

The new merged `PR` of `karmaWYSIWYD` (https://github.com/towardthesea/karmaWYSIWYD/pull/1) basically introduce a solution for the `curved movement` during `pushing`. Please kindly update your `karmaWYSIWYD` and test it if you can. 

Please also consider the `reachable region` problem of iCub for any further tests, e.g `behaviorManager`, my suggestion is not farther than (-0.3, -0.2) on the left side and (-0.3, 0.2) on the right side! Unluckily, in the `KARMA example` the `pushing tests` play with a little bit far objects. I will take the change of next branch to update them.

So I close the issue then!
Cheers,
Phuong

pattacini, 2016-12-01:
Let's keep this open until we'll test `karmaWYSIWYD` along with `IOL`.

Tobias-Fischer, 2016-12-07:
Shall we merge your branch then so we can close this issue? :)

pattacini, 2016-12-08:
Closing upon #180.

closed 2016-12-08 by pattacini