robotology-legacy/wysiwyd #170: Migrate from `karma` to new `karmaWYSIWYD`

towardthesea, 2016-11-16:
This issue is opened to provide a (painless) manual to migrate from original version of `karma` to new version, named `karmaWYSIWYD`.

**Where to get new version**
The modified forked version of `karma` is currently maintained in this repository: https://github.com/towardthesea/karmaWysiwyd. 

**How to build it**
You can simply compile it as you did with original `karma`. There is no new special requirement for it. If your system is working properly with `karma`, it will work with `karmaWYSIWYD`

**What is new**
- Fold the finger part of hand to form the `rake shape` in `pulling` action and provide possibility to tilt the whole hand to facilitate the `pulling` (through `handAngle` as below mentioned)
- New parameter `timeActions` to change the `time of motion`, higher the value leads to slower motions. It was tuned and carefully tested at `2.0s`. However it can be changed easily through `rpc` command as following:
  - `time set <timeActions(s)>`
- New parameter `handAngle` to change the `tilt angle` of the hand for `pulling`. It was tested and tuned at `15 deg` but can be changed through `rpc` command as:
  - `angl set <angle (deg) >`
- New parameter `safeMargin` to change the radius of the protected cylinder around the robot body so that the `pulling` action doesn't break the robot arm. Similarly, it is tuned and tested but can be changed by 'rpc' as:
  - `safe set <safeMargin (m)>`
- New `keepOtherArmSafe` method to raise the arm, which won't be used to execute actions, to keep it safe during `pushing` or `pulling`. By introducing this on, the `prepare` method in `subSystem_KARMA` will be removed (a warning comments about the situation were added in this update https://github.com/robotology/wysiwyd/commit/83a8e68ba2a984bd0945a90589f9e50f8c4815b7)

towardthesea, 2016-11-16:
Hi @pattacini , could you please kindly have a look. 
Thanks!

pattacini, 2016-11-17:
It seems fine, thanks @towardthesea 

cc @sockchinglow @clement-moulin-frier @jypuigbo

Tobias-Fischer, 2016-11-17:
Hi @towardthesea,
great, thanks! The only problem we have is that https://github.com/robotology/wysiwyd/tree/feature/upf/pushBehaviors is
a) unmerged
b) contains commits which introduce pushing behaviors in the `behaviorManager`, as well as changing the `subsystem_Karma`
c) is not compatible with the new `karmaWysiwyd`

To fix all of the above, I suggest to 
1) remove the `prepare` method in the `pushBehaviors` branch and use `keepOtherArmSafe` instead, 
2) then ask @sockchinglow to test whether the new behaviors nicely work with `karmaWysiwyd`, 
3) and then merge the `pushBehaviors` branch into `devel` ASAP (today?).

What do you think @pattacini?

Best,
Tobi

pattacini, 2016-11-17:
@Tobias-Fischer 

I've already talked to @towardthesea pointing out that next time we should follow a cleaner workflow and definitely avoid using branches not under the developer's own control.

As of now @towardthesea is at Humanoids and this will prevent him from reacting quickly.
Let's see when he'll have enough time to catch up with all these points.

towardthesea, 2016-11-17:
Hi @Tobias-Fischer , @pattacini 

Sorry for a bit delay of reply due to the time zone and conference.
1. Removing the `prepare()`: done
2. Testing the whole system: @sockchinglow could you please consider it without problem mentioned in https://github.com/robotology/wysiwyd/issues/168, because they can all be solved by adjusting `karmaWYSIWYD`. Please consider my first comment on this issue about the migration first.  Then we can move to PR step as soon as we can!

Cheers,
Phuong

Tobias-Fischer, 2016-11-23:
Can we close this as soon as #174 is merged?

towardthesea, 2016-11-23:
#174 is merged. This should be closed then!

closed 2016-11-23 by towardthesea

clement-moulin-frier, 2016-11-24:
Hi guys,

We're about to test it. Just to be sure: we just have to install the new version `karmaWYSIWYD` and all the rest remains the same, right?

towardthesea, 2016-11-24:
Hi @clement-moulin-frier , only new `karmaWYSIWYD` is required, the rest is the same. Remember to `make` and `make install`.