robotology-legacy/wysiwyd #172: Clean up / merge branches

Tobias-Fischer, 2016-11-17:
Hi guys,
I think we should try and reduce the number of branches before the integration meeting. Could you please give me an update on the status of these branches?
- `babblingTest`: no active development, not to be merged
- `narrative` @gregoire-pointeau can you create a pull request so we can merge this?
- `joints3d-to-ks2d` @maxime-petit what is the status on this? I can't remember whether we ended up using another methd
- `longTerm` I think we should merge this, it revives an old demo related to @maxime-petit and my TCDS paper. See https://github.com/robotology/wysiwyd/pull/171
- `r-4-learnPrimitive` Is this still in active development @maxime-petit? Or can it be merged?
- `samRelated`: See https://github.com/robotology/wysiwyd/pull/169
- `speech_actions`: Work in progress, see https://github.com/robotology/wysiwyd/pull/158
- `pushBehaviors`: I think we should merge this ASAP, see https://github.com/robotology/wysiwyd/issues/170
- `states`: Can you please let us know what that contains @jypuigbo? Can it be merged?
- `reactiveLayerEFAA`: Did you test this @clement-moulin-frier? Shall we merge? It only adds a new module and doesn't touch anything else, so I think it should be okay to merge.
- `renaming2`: Old and shouldn't be merged.

Thanks,
Tobi

gregoire-pointeau, 2016-11-17:
I'll clean my branch, reset nicely the conf files, and merge it.
It should be done before the end of the week.

maxime-petit, 2016-11-17:
`joints3d-to-ks2d` that was the first initial implementation of the `jointsAwareness` module. It was replicating some code for 3d joints position without the call to the cartesianSolver method. Ugo suggested to add that to `jointsAwareness` for performance issue. At the end we sticked with the original `jointsAwareness`, so I guess we can delete it.

`r-4-learnPrimitive` is still under development.

clement-moulin-frier, 2016-11-22:
Hi,
Regarding `reactiveLayerEFAA`, it contains code from EFAA about allostatic control with touch and speech interaction, as well as postures and face expressions.
It doesn't compile correctly due to changes in the subsystem interfaces. We spent a few hours a few weeks ago trying to adapt the code to the new interfaces, but it was not trivial (maybe @jypuigbo  remember better than me what was the main problems).
Nevertheless, I think we should keep this code available since it contains an interesting demo that we'll have to revive one day or another. 
How about keeping it in `modules/reactiveLayerEFAA`, without including that folder in the `CMakeLists.txt`? (as it is actually currently the case.)

Tobias-Fischer, 2016-11-22:
Hi, in this case I'd suggest not to merge it, but instead keep it in the separate branch. In case someone gets around to work on it, he can use the branch for it.
Is that fine with you?
Best, Tobi

Tobias-Fischer, 2016-11-24:
I've had another look at the `states` branch and I think it can be deleted. It contains an alternative way of propagating whether an object is in the reachable/non-reachable area, however we decided to go for a solution implemented in `iol2opc` instead. Can we delete @jypuigbo?

jypuigbo, 2016-11-28:
@Tobias-Fischer I'm checking it today and I let you know.

jypuigbo, 2016-11-29:
@Tobias-Fischer I't keep it by know, just temporarily. Because, do you think it's important to adapt the code you added in the `wrdac/knowledge` and `iol2opc` to have the kind of area definition I included in the branch `feature/upf/states` ? As a trade off, it increases flexibility for defining new setups, but reduces efficiency (see `app/reactiveLayer/sensationManager/conf/default.ini` and `OpcSensation::getState` in `src/modules/reactiveLayer/sensationManager/src/opcSensations.cpp` for reference). Let me know if you think it's worth adapting this change. 

In any of the cases, the relevant changes for planner/behaviors in that branch are now included in the [last commit](https://github.com/robotology/wysiwyd/commit/6c0acf724494ccdbe966cbabe4b403c550c01f3a) of the branch `feature/upf/pushingBehaviors` in order to test with `planner` and the `behaviorManager` as well.

Tobias-Fischer, 2016-11-30:
@jypuigbo okay thanks. Let's keep it then.

Tobias-Fischer, 2016-11-30:
Closing here as only one branch is left to be merged before the integration meeting.

closed 2016-11-30 by Tobias-Fischer