robotology-legacy/wysiwyd #194: Allow iCub pointing to object / body part somewhere in 3D space

Tobias-Fischer, 2016-12-13:
Hi,
We guys at Imperial (especially @hjchang) want to be able to have the iCub pointing to the humans body parts. More generally, you could think of having the iCub point anywhere in the space. The three points of elbow, hand, and object should be in a straight line.
Unfortunately, the `point()` provided by `ARE` is not suitable for this task (as the hand is always parallel to the ground).
@pattacini kindly offered to provide a method for this.

Thanks, Tobi

pattacini, 2016-12-13:
Further considerations:

- We can **block wrist joints** to `(0.0 0.0)`
- We can play with the **angle between the forearm and the arm** to handle the last degree of freedom

Tobias-Fischer, 2016-12-14:
@jypuigbo created a branch to add bodyparts with known kinect node to the sensation manager:
https://github.com/robotology/wysiwyd/tree/feature/sensationPointBodypart

Thanks!

Tobias-Fischer, 2016-12-20:
One thing which we make want to take into consideration as well is the `table_height` as defined by `ARE` ...

Tobias-Fischer, 2017-01-12:
@pattacini: Just a quick reminder :)

pattacini, 2017-01-12:
Yeah 😉 I've planned to start working this out from this weekend.
Keep you informed.

Tobias-Fischer, 2017-01-12:
Great, thanks a lot :)

pattacini, 2017-01-19:
Hi @Tobias-Fischer 

The new pointing is implemented as per https://github.com/robotology/icub-main/pull/413.

What is still missing is to expose it from within `wysiwyd::wrdac::SubSystem_ARE`. Could you do that? I'll be busy with other issues next days, while hoping the new feature will be working always fine...

Tobias-Fischer, 2017-01-20:
Thanks a lot! We'll give it a try in the next few days. No worries about the `SubSystem`, I'll do that.
Best, Tobi

Tobias-Fischer, 2017-01-20:
@hjchang Are you free on Monday? Let's try this.

Tobias-Fischer, 2017-01-27:
Hi @pattacini, as mentioned on slack it seems that ARE never returns after sending `pfar`. We'll close this issue as soon as this is fixed. Thanks again a lot for implementing it, the pointing itself is nice and smooth.

pattacini, 2017-01-27:
Hi @Tobias-Fischer 

I've relaxed the requirement for reaching the target (https://github.com/robotology/icub-main/commit/a106d3dcb1ce9d70f22fe08a112de1cbaf02bd65).
Hope this helps.

Tobias-Fischer, 2017-02-06:
Hi @pattacini,
Do you think you can apply this fix on `icub-main`, which for some reason is needed on `iCubLondon01` for the method to return?
```
-        iarm->waitMotionDone();
+        iarm->waitMotionDone(0.1,5.0);
```
Once that's fixed we can close this issue :).

pattacini, 2017-02-06:
Only because you're my friend 😉 
Done in https://github.com/robotology/icub-main/commit/0b49fc997d4b8df05bddd0265f97977f204ce913.

Tobias-Fischer, 2017-02-06:
:)
Thanks a lot!

closed 2017-02-06 by Tobias-Fischer