robotology-legacy/wysiwyd #205: SAMErrors

clement-moulin-frier, 2016-12-15:
Hi @dcam0050 

We are running the demo and we found the error in the attached file. 
[SAMERROR.txt](https://github.com/robotology/wysiwyd/files/654701/SAMERROR.txt)

Thanks

Tobias-Fischer, 2016-12-15:
Also, action recognition did not work (it never replied, and it opened a new window when sending the recognition order). @dcam0050 please be in touch with the guys from UPF such that the action + face recognition works nicely for the review. We can't afford not getting a reply from SAM. There should be a hard timeout of 10 seconds or so; if it can't load the model in this time, it should reply "nack".

Tobias-Fischer, 2016-12-21:
Also a pointer that `pip install tables` is needed is missing.

dcam0050, 2016-12-23:
The last commit should have fixed the mentioned issues. I've also added a logging function that will capture all exceptions and writes them to a log file within the data folder specified in default.ini. Also the non responsive countdown does not apply to models that are being loaded right now so they can take as much time as needed without the risk of being restarted continuously. Hope this makes it more stable

Tobias-Fischer, 2017-01-05:
Hi @clement-moulin-frier @jypuigbo,
Can you please try whether SAM is working fine in Barcelona with the latest changes? In this case, we can close this issue.