robotology-legacy/wysiwyd #210: Fix bugs in the planner

clement-moulin-frier, 2017-01-18:
Cc @sockchinglow @jypuigbo 

The planner module is still not behaving as expected in some situations. The following problems sometime occur:

- When a drive is close to the threshold and the human give an order, the execution of the plan and of the drive-related behavior can be mixed together (e.g. 1st action of the plan, then drive-related behavior, then 2nd action of the plan). A temporary solution for avoiding it is to not giving orders to the robot when a drive is close to the threshold.
- The order `point to the X` when `X` is an unknown object is supposed to ask the human to point to `X`. We have observed that this sometimes lead to a blocking situation. A temporary solution is to avoid asking the robot to point to an unknown object, except if this object is the only unknown one (in that case the robot will deduct the object, this works fine). The reason for this bug could be related to the previous point.

I will sit with @sockchinglow early next week to debug the planner and we'll keep you updated on the progress here.

sockchinglow, 2017-01-25:
Repetitive error message describing the failure to execute a plan is addressed in #212

sockchinglow, 2017-01-25:
Occasionally, within the time for a command to be processed and sent to the `planner`, the `allostaticController` starts another behaviour in the `behaviorManager` based on the internal drive dynamics. This causes the `planner` to send the command to the `BM` and receiving a `nack` response. As this response was included in the attempt count, which has a limit of 3, the planner interjects the ongoing action with the error message (something like "I have tried too many times etc.").

This is rectified by f8917d525b6989ec96020c17be21dbf45a95cf3e where now, once `planner` receives a `nack` from the `BM` it cancels the plan instead of reattempting the plan. It says "wait a second" for a verbal indication of this (this can, of course, be changed to be much more informative).

clement-moulin-frier, 2017-01-25:
We still need to check and test it on the robot though, we'll do it tomorrow Thursday

Tobias-Fischer, 2017-01-26:
So what does that mean for the plan to be executed though? Say the plan consists of behaviors B1 and B2 (e.g. B1: Ask human to push the object to shared area, B2: Pull object). The `planner` sends the command for B1 to `BM` which is successfully executed. Now `allostaticController` kicks in and asks `BM` to execute B3. `planner` tries to get B2 executed, but `BM` replies `nack` as B3 is currently executed. The `planner` gives up, the plan fails, and B1+B3 were randomly executed rather than the meaningful B1+B2. Is that right?

Wouldn't it be better if the `planner` asks `allostaticController` to not do anything while the plan is being executed?

Best, Tobi

clement-moulin-frier, 2017-01-26:
That is actually the case, the drives are frozen during the entire plan execution. The change is that now the planner give up if the 1st action can't be executed because the BM is busy. Before it was keeping trying, postponing the plan execution. This was noy good for the interaction because the plan was often executed long after it was requested.

Tobias-Fischer, 2017-01-26:
Okay, I see, this makes sense. However, shouldn't the fix be on a higher level then? Shouldn't there be a "common mutex" which only allows access to the BM by either the `allostaticController` or the `planner`?

clement-moulin-frier, 2017-01-26:
This is also the case, there is a [mutex in the BM](https://github.com/robotology/wysiwyd/blob/devel/main/src/modules/reactiveLayer/behaviorManager/include/behavior.h#L40-L61) for this.

sockchinglow, 2017-01-26:
@Tobias-Fischer the mutex in the `BM` successfully prevents the planner from sending commands to it while it is running another behaviour. However, because the planner used to take this as a failure of action execution it was added into the same attempt count as sensing the failure of an action (e.g. seeing that the object has not changed position when it should have).

Tobias-Fischer, 2017-01-26:
Okay cool .. seems like you guys found the bug :). Did you get to test it yet?

sockchinglow, 2017-01-26:
Not yet. I'm also working on #212 so that the error sentence is more natural.

jypuigbo, 2017-01-26:
Tested. 
- `nack`from `BM` is no longer a problem. Planner has been sped up and if there is a conflict instead of overlapping it just reports it. 
- Fixed the bug about not meeting preconditions
- The 3rd error was a bug in our grammar

I keep it open as we are still improving it slightly. More mews coming soon

Tobias-Fischer, 2017-01-26:
Well done, sounds good. Did you save a log file by any chance?

clement-moulin-frier, 2017-01-27:
The three last commits in `fix/planner` allow a better checking of when a plan can be executed or not. It has been tested without the robot. We'll test it on the full demo on Monday.

clement-moulin-frier, 2017-01-30:
Hi,

The last commits on this branch have been tested and the robot and everything works fine here at UPF. Can another lab give it a try? Then I think we can merge.

Cheers,

Tobias-Fischer, 2017-02-02:
Closing here, let's open another issue if new bugs show up.

closed 2017-02-02 by Tobias-Fischer