robotology-legacy/wysiwyd #212: When asking to point an unknown object, the robot repeats 3 times the same thing.

clement-moulin-frier, 2017-01-25:
When asking to point an unknown object, the robot says sonthing like:

"I cannot execute the plan because the object is not known and not present and not known and not present and not known and not present".

@sockchinglow knows how to solve it

jypuigbo, 2017-01-25:
The issue has been tracked to planner. 
Log from planner: 
```
104174	1580,078122	DEBUG	Command received:  manual on
104175	1580,078133	DEBUG	Now manual mode is  true
105508	1588,523583	DEBUG	Command received:  new (move 1 (bodypart little))
105509	1588,523621	DEBUG	checking priority of cmd
105510	1588,523635	INFO	Plan move is not known. Use listplans to see available plans.
109894	1617,171937	DEBUG	Command received:  new (move 1 (bodypart middle))
109895	1617,171949	DEBUG	checking priority of cmd
109897	1617,171969	INFO	Plan move is not known. Use listplans to see available plans.
112053	1632,911083	DEBUG	Command received:  new (move 1 (bodypart little))
112054	1632,911095	DEBUG	checking priority of cmd
112055	1632,911106	INFO	Plan move is not known. Use listplans to see available plans.
114259	1648,780763	DEBUG	Command received:  new (take 1 (object octopus))
114260	1648,780778	DEBUG	checking priority of cmd
114261	1648,780791	INFO	plan is known: take
114263	1648,780814	INFO	command to be added to newPlan:  new (take 1 (object octopus))
114375	1649,431251	DEBUG	moving plan from newPlan to action_list
114378	1649,431304	DEBUG	indiv from rep is  false
114379	1649,431314	DEBUG	ultimate goal is already met:  false
114381	1649,431339	INFO	there are preconditions to check for.
114384	1649,431373	DEBUG	indiv from rep is  false
114385	1649,431380	DEBUG	lArgument formed for condition
114388	1649,431411	DEBUG	indiv from rep is  false
114390	1649,431425	DEBUG	lArgument formed for condition
114392	1649,431446	DEBUG	indiv from rep is  false
114394	1649,431456	DEBUG	lArgument formed for condition
114401	1649,431492	DEBUG	ABM is connected, recording...
114416	1649,574782	INFO	moveObject reasoning has been recorded in the ABM
114417	1649,574819	INFO	Action is ask/moveObject, objectType now describes if desired action.
114418	1649,574833	INFO	direction:  back
114419	1649,574862	INFO	there are preconditions to check for.
114423	1649,574910	DEBUG	indiv from rep is  false
114425	1649,574932	DEBUG	lArgument formed for condition
114429	1649,574976	DEBUG	indiv from rep is  false
114431	1649,574997	DEBUG	lArgument formed for condition
114434	1649,575030	DEBUG	indiv from rep is  false
114436	1649,575051	DEBUG	lArgument formed for condition
114437	1649,575061	DEBUG	ABM is connected, recording...
114442	1649,635545	INFO	ask reasoning has been recorded in the ABM
114443	1649,635562	INFO	Action is ask/moveObject, objectType now describes if desired action.
114444	1649,635575	INFO	direction:  push
114445	1649,635586	WARNING	None of the sets of prerequisites are met, unable to handle plan.
114446	1649,635598	WARNING	No partner present was found!
114447	1649,635609	WARNING	[iCubClient] Called look() on an unavailable target: "  "
114448	1649,635619	INFO	I could not execute the plan take because the octopus is not present and not known and not present and not known
114449	1649,635632	DEBUG	iCub says I could not execute the plan take because the octopus is not present and not known and not present and not known
114452	1649,635666	DEBUG	addressee is null, getting present agent.
114453	1649,635677	DEBUG	addressee is now partner
123605	1712,417003	DEBUG	Command received:  new (take 1 (object octopus))
123606	1712,417014	DEBUG	checking priority of cmd
123607	1712,417025	INFO	plan is known: take
123608	1712,417036	INFO	command to be added to newPlan:  new (take 1 (object octopus))
123706	1712,852399	DEBUG	moving plan from newPlan to action_list
123708	1712,852453	DEBUG	indiv from rep is  false
123710	1712,852472	DEBUG	ultimate goal is already met:  false
123712	1712,852489	INFO	there are preconditions to check for.
123715	1712,852511	DEBUG	indiv from rep is  false
123717	1712,852526	DEBUG	lArgument formed for condition
123719	1712,852540	DEBUG	indiv from rep is  false
123722	1712,852566	DEBUG	lArgument formed for condition
123731	1712,852633	DEBUG	indiv from rep is  false
123732	1712,852640	DEBUG	lArgument formed for condition
123733	1712,862532	DEBUG	ABM is connected, recording...
123736	1712,903308	INFO	moveObject reasoning has been recorded in the ABM
123737	1712,903351	INFO	Action is ask/moveObject, objectType now describes if desired action.
123740	1712,903378	INFO	direction:  back
123742	1712,903394	INFO	there are preconditions to check for.
123743	1712,903401	DEBUG	indiv from rep is  false
123745	1712,903421	DEBUG	lArgument formed for condition
123747	1712,903435	DEBUG	indiv from rep is  false
123748	1712,903443	DEBUG	lArgument formed for condition
123749	1712,903451	DEBUG	indiv from rep is  false
123751	1712,903466	DEBUG	lArgument formed for condition
123757	1712,903509	DEBUG	ABM is connected, recording...
123764	1712,943574	INFO	ask reasoning has been recorded in the ABM
123765	1712,943592	INFO	Action is ask/moveObject, objectType now describes if desired action.
123766	1712,943606	INFO	direction:  push
123767	1712,943617	WARNING	None of the sets of prerequisites are met, unable to handle plan.
123768	1712,943636	WARNING	No partner present was found!
123769	1712,943649	WARNING	[iCubClient] Called look() on an unavailable target: "  "
123770	1712,943661	INFO	I could not execute the plan take because the octopus is not present and not known and not present and not known
123771	1712,943675	DEBUG	iCub says I could not execute the plan take because the octopus is not present and not known and not present and not known
123773	1712,953597	DEBUG	addressee is null, getting present agent.
123774	1712,953632	DEBUG	addressee is now partner
```

sockchinglow, 2017-01-26:
The repetition of conditions is fixed in 408b2cd479708263a520e33ae314590aea80c47a. Additionally, the iCub will not mention anything about location of an object that is not present. It will also not mention anything about presence or location if the object is not known.

jypuigbo, 2017-01-26:
All this has been tested this far and it's solved.

closed 2017-01-26 by jypuigbo