robotology-legacy/wysiwyd #213: Pulling an object too close, preventing correct detection

clement-moulin-frier, 2017-01-30:
Hi,

When executing the plan `take`, often the robot pull the object too close to itself, preventing it to be correctly detected when the post-condition is checked. In consequence the plan fails.

Is it possible to modify the `pull` behavior  so that the object remains in the visible area?

cc @towardthesea @pattacini @jypuigbo @sockchinglow

Tobias-Fischer, 2017-01-30:
You can change the `backPos` of the `default.ini` of the `BM`. Just set it to a more negative value and give it a try. Best, Tobi

towardthesea, 2017-01-30:
Thanks @Tobias-Fischer :smile:

clement-moulin-frier, 2017-01-30:
Thanks guys (I remember one already told me this but I forgot ...)
It works much better with `backPos` to `-0.40`.
Should I push the change in `default.ini` or do we consider it as specific to the UPF setup?

Tobias-Fischer, 2017-01-30:
You can push the change to the repo :). Closing as the issue seems resolved.

closed 2017-01-30 by Tobias-Fischer