robotology-legacy/wysiwyd #215: proactiveTagging sometimes tag an object with the label "unknown"

clement-moulin-frier, 2017-02-02:
When proactively tagging an object, sometimes the robot says "I get it, this is an unknown". (whereas the human said e.g. "this is a duck").

The problem seems to comes from [here](https://github.com/robotology/wysiwyd/blob/devel/main/src/modules/attentionRelated/proactiveTagging/src/proactiveTagging.cpp#L409-L421), where "unknown" is the default value if there is no "object role" in the bottle returned by the recognizer.

I see two possible ways of solving it:

1) Understanding why there is no "object role" in that bottle, whereas it is [not optional in the grammar](https://github.com/robotology/wysiwyd/blob/devel/main/app/attentionRelated/proactiveTagging/conf/GrammarAskNameObject.xml#L21)

2) Checking in `proactiveTagging` if "unknown" has been returned and in that case not updating the OPC and saying "I didn't understand".

For the second option, where would be the best place in `proactiveTagging` to do it? (cc @Tobias-Fischer @maxime-petit ). But understanding point 1) would be better :)

Clément

jypuigbo, 2017-02-03:
We found another related bug this morning:

When the robot asked "What is this object?", then the human said "This is the duck", the robot eventually replied "I asked you something else".

Looking at the logs, we see that speech recognizer actually returned:
```
139074	1322,105474	INFO	 Reply from Speech Recog :  ACK ("Let us talk about pong" (SUBNODE (ABOUT (keyword pong))))
```

i.e., something from the main grammar.

Looking at the code, the call of speech recognizer in practiveTagging occurs [here](https://github.com/robotology/wysiwyd/blob/devel/main/src/modules/attentionRelated/proactiveTagging/src/proactiveTagging.cpp#L358-L368)

i.e. specifying the grammar `GrammarAskNameObject` as the argument of `iCub->getRecogClient()->recogFromGrammarLoop`.

So at the end it seems to be a similar problem as in the previous post: speech recognizer is called with a particular grammar, but returns something from another one.

Who knows this module well enough to help here? Maybe @maxime-petit ?

Thanks,
Clément

Tobias-Fischer, 2017-02-03:
Hi guys,
A few notes:
1) The `speechRecognizer` shouldn't return something from the default grammar. Are you guys sure you are muting the microphone properly so no noise is detected? Is the `speechRecognizer` up-to-date?
2) `proactiveTagging` is looking checking against an `error` string rather than a `unknown` string. I changed this here: https://github.com/robotology/wysiwyd/commit/e5f6e00a99f3127a6fa0f1009ffeac8da785efbb so now the robot says "I asked you something else", and the object is not being renamed. Obviously as you said this is not the perfect fix, as you will need to trigger the interaction again.

How often does this happen?

Best, Tobi

Tobias-Fischer, 2017-02-09:
Proper fix in https://github.com/robotology/wysiwyd/commit/1067d58ec32fdde74b2824e14d883a2d98ad815e - let's close this.

commit 1067d58e (2017-02-09): Fix #215 - Loop if wrong grammar was returned
closed 2017-02-09 by Tobias-Fischer