robotology-legacy/wysiwyd #217: Change the NarrativesHandler text to use our names and our objects

jypuigbo, 2017-02-06:
@Tobias-Fischer pointed out we should do the changes in the `opcPopulater`. Mainly is changing `Allan` by some of our names and making sure that the object names are the ones we'll be using for the demo. 

Specifically, the object names we've been using since the last integration has been `blue cube`, `octopus` and `duck`, and, sometimes, `nemo` or `fish`.

Tobias-Fischer, 2017-02-06:
Yes, that's correct. The names need to be changed in the methods `populateSpecificX`, and afterwards the procedure described in https://github.com/robotology/wysiwyd/tree/devel/main/src/modules/narrativeRelated/narrativeGraph/src needs to be followed (i.e. updating the instance numbers for the start and end of the narration). It should be straightforward :).

PS: When following the procedure, I'd recommend doing this when the robot is turned off. Otherwise, the robot will move, so be careful.

Tobias-Fischer, 2017-02-09:
Partly fixed with https://github.com/robotology/wysiwyd/commit/e805eb5e1c5ef52b4fc29f6f6eee9c52c5dce804
We now need to populate new instances using `opcPopulater` and change the config file of `narrativeGraph`.

Tobias-Fischer, 2017-02-09:
Done!

closed 2017-02-09 by Tobias-Fischer