robotology-legacy/wysiwyd #219: Use delayed stop for modules within yarpmanager

Tobias-Fischer, 2017-02-06:
Thanks to https://github.com/robotology/yarp/issues/1051 we hopefully will be able to stop all modules within the WYSIWYD year 3 XML gracefully. We will need to figure out which order is best to stop the modules. I can take care of this. Anything I should take into consideration @pattacini?

pattacini, 2017-02-06:
Take a look at here: https://github.com/robotology/icub-main/commit/1fc3101876393ebde44c38bcc190e6376ec35565

Low level modules should be stopped with a delay in order to give higher level modules enough time to close resources gracefully.

In our settings low level components can be:
- `iKinGazeCtrl`
- `OPC`

Tobias-Fischer, 2017-02-09:
Done. Let's see whether we found the good order :)

Tobias-Fischer, 2017-02-09:
This works amazingly well. The new config file was added here: https://github.com/robotology/wysiwyd/commit/67cfcb64985e256c9e956cd07ee3df77574c3575
Thanks so much @apaikan :)

closed 2017-02-09 by Tobias-Fischer