robotology-legacy/wysiwyd #220: Torso shaking when pulling an object

clement-moulin-frier, 2017-02-08:
Hi,

We just observed a weird behavior. We asked the robot to pull an object, it said "I will pull the object to the back", initialized the movement, and started shaking the torso badly (we had to push the red button).
Looking at the logs, we saw that the object perceived position was at -0.36, whereas the `backPos` in BM is actually at -0.40. So the object was actually localized between the target position and the robot. This was not the case in reality and it might be due to some error in the localization process, so we should probably recalibrate.
The question is: does Karma deals with this kind of situation where we ask to pull an object which is perceived closer to the robot than it should be? And can this produce the torso shaking we have observed? If so, it will be useful to handle this particular situation and cancel the pull action (or the push action in in the reverse situation).

Thanks,
Clément
Cc @towardthesea @pattacini

jypuigbo, 2017-02-09:
Because of the recurrent shakings, should we also revise, just in case, that the PIDs and currents are correct? referring to the previous problems in [icub-support#30](https://github.com/robotology/icub-support/issues/30)
@pattacini

Tobias-Fischer, 2017-02-11:
Closing here as this didn't happen for a while.

closed 2017-02-11 by Tobias-Fischer